schreibfaul1/ESP32-audioI2S #630: audio.connecttospeech() function not working on long sentences

DuyenCdt, 2023-12-20:
I get a memory error message when using a long sentence in the audio.connecttospeech() function, is there any way to save that sentence to SD memory instead of Esp32 memory, or is there another way to save it to free memory on Esp32?
Thank you
[image]

spikepavel, 2024-01-06:
Even if you have a lot more memory, the long message won't work. This is some kind of error and apparently an additional check for buffer overflow is needed. This is observed in many people. I split the text into lines and send it in batches, otherwise an error reboot occurs.

schreibfaul1, 2024-01-06:
Google limits the length of the string. I don't know of any equivalent free alternative.

aber68, 2024-12-17:
hello, how this "batch" may look alike? sending sentences one after the other will only play the last one ... br andreas

spikepavel, 2024-12-17:
Hello. If I understood you correctly, then you need to pause equal to the playback time of the last send. I did it so that I counted the number of letters and made a final delay for each letter before sending the next sentence.

aber68, 2024-12-18:
i tried several attempts with audio_eof_stream, but it always brings the last chunked sentence. i have no idea how to put out a length >100. do you have a code example which will do the job even with truncated chunks of phrases?

spikepavel, 2024-12-18:
No way. Break it down into several words. Wait for the end of playback and submit a new batch of words to the input. It uses the free version of Google TTS. If you want large packages (25 thousand characters), then only for money.

aber68, 2024-12-18:
what is the length limit of Google TTS free? i estimate around 100 ...

hammerheaddown, 2025-02-06:
so does anyone know the length of words or characters that we can use with google?

i like the idea @spikepavel   mentioned.  if he would tell us more detail on how he did it that would be great.

if you don't want to share your code, maybe let us know what is the sweet spot  of where to break up the sentences