schreibfaul1/ESP32-audioI2S #631: Seeed XIAO ESP32s3 reboots when trying to play Pink Panther wav file

Les-A, 2023-12-21:
Was working fine with older library. Downloaded and installed the latest version yesterday to get OGG-Vorbis module. Sadly, Vorbis needs psram which this device doesn't have but in the going I discovered this problem while trying to play mp3 and wav files using urls (have no need of SD Cards).  mp3 files play just fine.  Captured the following:

> 14:40:33.291 -> Test Source
14:40:33.291 -> info        Connect to new host: "https://github.com/schreibfaul1/ESP32-audioI2S/raw/master/additional_info/Testfiles/Pink-Panther.wav"
14:40:33.291 -> info        buffers freed, free Heap: 210856 bytes
14:40:34.320 -> info        SSL has been established in 1015 ms, free Heap: 173988 bytes
14:40:34.320 -> Webserver url args: 0
14:40:34.320 -> 
14:40:34.718 -> info        redirect to new host "https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Pink-Panther.wav"
14:40:34.718 -> info        Connect to new host: "https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Pink-Panther.wav"
14:40:34.718 -> info        buffers freed, free Heap: 211012 bytes
14:40:35.281 -> info        SSL has been established in 550 ms, free Heap: 175048 bytes
14:40:35.646 -> info        Filename is additional_info/Testfiles/Pink-Panther.wav
14:40:35.646 -> lasthost    https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Pink-Panther.wav
14:40:35.745 -> info        stream ready, buffer filled in 90 ms
14:40:35.745 -> info        FormatCode: 1
14:40:35.745 -> info        DataRate: 88200
14:40:35.745 -> info        DataBlockSize: 4
14:40:35.745 -> info        BitsPerSample: 16
14:40:35.745 -> info        Audio-Length: 621484
14:40:35.745 -> info        syncword found at pos 0
14:40:35.745 -> info        Channels: 2
14:40:35.745 -> info        SampleRate: 22050
14:40:35.745 -> info        BitsPerSample: 16
14:40:35.745 -> info        BitRate: 705600
14:40:35.978 -> 
14:40:35.978 -> assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)
14:40:35.978 -> 
14:40:35.978 -> 
14:40:35.978 -> Backtrace: 0x40377342:0x3fcebb00 0x4037cafd:0x3fcebb20 0x403829ad:0x3fcebb40 0x4037dae6:0x3fcebc70 0x4204f45e:0x3fcebcb0 0x4203d101:0x3fcebcd0 0x4203d866:0x3fcebcf0 0x4203d882:0x3fcebd10 0x4203ea1c:0x3fcebd30 0x4203eae2:0x3fcebd90 0x42034c75:0x3fcebdb0 0x420a545e:0x3fcebdd0 0x42060815:0x3fcebdf0 0x4205d741:0x3fcebe10 0x4205df6b:0x3fcebe30 0x4205e6ae:0x3fcebea0 0x42023bf2:0x3fcebec0 0x42023231:0x3fcebee0 0x42010acd:0x3fcebf00 0x42011939:0x3fcebf30 0x42003a58:0x3fcebf50 0x42029db5:0x3fcebf70
14:40:36.211 ->

schreibfaul1, 2023-12-22:
I can't understand that, https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Pink-Panther.wav runs for me without any problems. Based on the log, I suspect a timing problem (collision?) between two tasks?

Les-A, 2023-12-23:
Hi. Thanks for looking at this for me. I got the same results with an ESP-WROOM-32 (again with no psram). I tried the "Simple_Wifi_Radio" example set up to play the Pink Panther test file and it failed with the WROOM.  Here is the slightly different output ...

> 2023-12-23 23:16:21 b'Rebooting...\r\n'
2023-12-23 23:16:21 b'ets Jun  8 2016 00:22:57\r\n'
2023-12-23 23:16:21 b'\r\n'
2023-12-23 23:16:21 b'rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)\r\n'
2023-12-23 23:16:21 b'configsip: 0, SPIWP:0xee\r\n'
2023-12-23 23:16:21 b'clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00\r\n'
2023-12-23 23:16:21 b'mode:DIO, clock div:1\r\n'
2023-12-23 23:16:21 b'load:0x3fff0030,len:1184\r\n'
2023-12-23 23:16:21 b'load:0x40078000,len:13260\r\n'
2023-12-23 23:16:21 b'load:0x40080400,len:3028\r\n'
2023-12-23 23:16:21 b'entry 0x400805e4\r\n'
2023-12-23 23:16:21 b'.audio_info: Connect to new host: "https://github.com/schreibfaul1/ESP32-audioI2S/raw/master/additional_info/Testfiles/Pink-Panther.wav"\r\n'
2023-12-23 23:16:24 b'audio_info: PSRAM not found, inputBufferSize: 14399 bytes\r\n'
2023-12-23 23:16:24 b'audio_info: buffers freed, free Heap: 157332 bytes\r\n'
2023-12-23 23:16:24 b'audio_info: SSL has been established in 1250 ms, free Heap: 116712 bytes\r\n'
2023-12-23 23:16:25 b'10\r\n'
2023-12-23 23:16:25 b'0\r\n'
2023-12-23 23:16:25 b'audio_info: redirect to new host "https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Pink-Panther.wav"\r\n'
2023-12-23 23:16:25 b'audio_info: Connect to new host: "https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Pink-Panther.wav"\r\n'
2023-12-23 23:16:25 b'audio_info: buffers freed, free Heap: 156508 bytes\r\n'
2023-12-23 23:16:25 b'audio_info: SSL has been established in 632 ms, free Heap: 115032 bytes\r\n'
2023-12-23 23:16:26 b'audio_info: Filename is additional_info/Testfiles/Pink-Panther.wav\r\n'
2023-12-23 23:16:26 b'audio_info: stream ready, buffer filled in 71 ms\r\n'
2023-12-23 23:16:26 b'audio_info: FormatCode: 1\r\n'
2023-12-23 23:16:26 b'audio_info: DataRate: 88200\r\n'
2023-12-23 23:16:26 b'audio_info: DataBlockSize: 4\r\n'
2023-12-23 23:16:26 b'audio_info: BitsPerSample: 16\r\n'
2023-12-23 23:16:26 b'audio_info: Audio-Length: 621484\r\n'
2023-12-23 23:16:26 b'audio_info: syncword found at pos 0\r\n'
2023-12-23 23:16:26 b'audio_info: Channels: 2\r\n'
2023-12-23 23:16:26 b'audio_info: SampleRate: 22050\r\n'
2023-12-23 23:16:26 b'audio_info: BitsPerSample: 16\r\n'
2023-12-23 23:16:26 b'audio_info: BitRate: 705600\r\n'
2023-12-23 23:16:26 b'CORRUPT HEAP: Bad head at 0x3ffe854c. Expected 0xabba1234 got 0xfcdefb7a\r\n'
2023-12-23 23:16:26 b'\r\n'
2023-12-23 23:16:26 b'assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)\r\n'
2023-12-23 23:16:26 b'\r\n'
2023-12-23 23:16:26 b'\r\n'
2023-12-23 23:16:26 b'Backtrace: 0x4008375d:0x3ffd6b60 0x4008bbf1:0x3ffd6b80 0x400910c5:0x3ffd6ba0 0x40090cbd:0x3ffd6cd0 0x40083de9:0x3ffd6cf0 0x40085a05:0x3ffd6d10 0x4013f3ae:0x3ffd6d30 0x40130b2a:0x3ffd6d50 0x400f29ef:0x3ffd6d70 0x400f1fac:0x3ffd6d90 0x400f208d:0x3ffd6db0 0x400dfc51:0x3ffd6dd0 0x400e0a85:0x3ffd6e00 0x400d2f9e:0x3ffd6e20 0x400f5685:0x3ffd6e40\r\n'
2023-12-23 23:16:26 b'\r\n'

I tried to set up the ESP32S3 for a similar test but it required too many changes to get it to work as I don't have the extra hardware and in the end it broke and wouldn't boot up so I gave up.

As I can do without the wav capability and can live with mp3 only I'm happy to close this as I have neither the time nor technical knowledge to take it any further.

Thanks again for the response, and a very useful library.