schreibfaul1/ESP32-audioI2S #643: Can't play audio from ESP32-WROOM-32

ThePortugueseMan, 2023-12-30:
I can't get my ESP32-WROOM-32 DEV board to play audio. I'm using the MAX98357 as a DAC connected to a single 4 Ohm 3 Watt speaker.

I'm not sure what the problem could even be, but here is what I tried:
- MAX98357 + Speaker (4 Ohm 3W)
Followed this [Adafruit tutorial](https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp/raspberry-pi-test), and the amplifier + speaker is working correctly with a Raspberry Pi Zero 2W
Tried using an external 5V to power the Amp

- Connections
Moved from breadboard to female-female DuPont connectors to reduce noise
Checked a thousand times if the pins were connected correctly

- Software
Tried Platformio and Arduino IDE (same sketches)
Tried web streams, web files and SD card
I was initially working with v2, and have now updated to v3
Tried different sketches (from other issues, wiki and ReadMe)

- Hardware
Checked for continuity between the labeled PCB pins and the ESP's pins to see if they matched
Tested other sketches (not I2S/audio related) to see if the board itself was faulty 

- Related info
I was having the problem described in this issue #631 . As described, when playing a .wav file it crashed, so now I'm using .mp3

Here's my code:

```
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
#define SD_CS          5
#define SPI_MOSI      23
#define SPI_MISO      19
#define SPI_SCK       18

#define I2S_DOUT      26  // connect to DAC pin DIN
#define I2S_BCLK      27  // connect to DAC pin BCK
#define I2S_LRC       25  // connect to DAC pin LCK

#define LED_ONBOARD 2
const long interval = 1000;
int ledState = LOW; 
unsigned long previousMillis = 0; 

Audio audio;

String ssid =     "****";
String password = "****";

void setup() {
    
    // pinMode(SD_CS, OUTPUT);      digitalWrite(SD_CS, HIGH);
    // SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    // SD.begin(SD_CS);
    Serial.begin(115200);

    pinMode(LED_ONBOARD,OUTPUT);
    WiFi.disconnect();
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid.c_str(), password.c_str());
    while (WiFi.status() != WL_CONNECTED) delay(1500);
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21); // default 0...21
    audio.forceMono(true);

//  *** web files ***
 audio.connecttohost("https://github.com/schreibfaul1/ESP32-audioI2S/raw/master/additional_info/Testfiles/Olsen-Banden.mp3");

}

void loop()
{

   unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;

    if (ledState == LOW) {
      ledState = HIGH;
    } else {
      ledState = LOW;
    }

    digitalWrite(LED_ONBOARD, ledState);
  }
  audio.loop();
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
    Serial.print("eof_speech  ");Serial.println(info);
}
```

And the output:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
info        Connect to new host: "https://github.com/schreibfaul1/ESP32-audioI2S/raw/master/additional_info/Testfiles/Olsen-Banden.mp3"
info        PSRAM not found, inputBufferSize: 14399 bytes
info        buffers freed, free Heap: 194356 bytes
info        SSL has been established in 1225 ms, free Heap: 153536 bytes
info        redirect to new host "https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Olsen-Banden.mp3"
info        Connect to new host: "https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Olsen-Banden.mp3"
info        buffers freed, free Heap: 193160 bytes
info        SSL has been established in 608 ms, free Heap: 151916 bytes
info        Filename is additional_info/Testfiles/Olsen-Banden.mp3
info        MP3Decoder has been initialized, free Heap: 125580 bytes , free stack 2684 DWORDs
lasthost    https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Olsen-Banden.mp3
info        stream ready, buffer filled in 66 ms
info        ID3 framesSize: 265
info        ID3 version: 2.4
info        ID3 extended header
id3data     Year: 2014
id3data     Album: Olsen-Banden
id3data     Artist: Bjarne Liller - Olsen Banden (Titelmusik der Olsenbande)
info        Audio-Length: 328698
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: 128000
info        VBR recognized, audioFileDuration is estimated
info        End of webstream: "https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Olsen-Banden.mp3"
```

I'm all out of ideas, so any help would be greatly appreciated!

schreibfaul1, 2023-12-30:
It looks good so far, I can't see any errors at first glance. You can use (almost) any GPIO for the DAC. Try three others. Maybe one pin has contact problems.

ThePortugueseMan, 2023-12-30:
I knew there was something left to try... I was always using some combination of those 3 pins, seems like one has an issue, I'll have to figure out what it is.

Thanks for the help!

closed 2023-12-30 by ThePortugueseMan