schreibfaul1/ESP32-audioI2S #644: schematic shows 5v connected to PCM5102 board but limit is 3.9v

wfdudley, 2024-01-02:
This image:
ESP32-audioI2S/additional_info/ESP32_I2S_PCM5102A.JPG

shows 5 volts connected to the "VIN" pin on the PCM5102 board,
however, the Texas Instruments data sheet says the Vcc supply should be 3.3v, and the maximum
is 3.9v.

Thanks.

schreibfaul1, 2024-01-02:
It is essential to ensure that the maximum operating voltage is not exceeded. However, the PCM5102 board shown here, like many others, has an internal voltage regulator that limits the voltage to 3.3V.
[image]

wfdudley, 2024-01-02:
RIght you are, sorry to waste your time.
Thanks for your work on this project.

Bill Dudley
This email is free of malware because I run Linux.