schreibfaul1/ESP32-audioI2S #645: 'Hostaddress is empty' trying to play BBC .m3u8 stream

Alkerr, 2024-01-02:
Hello, love the library, thanks for sharing it. I'm having trouble playing the latest BBC streams. It gets the first playlist and plays the .ts files OK but then it just stops. I've tried calling connecttohost again when isRunning is false but there is either repetition or loss when that happens.

What am I doing wrong?

Thanks!

I'm using an ESP32_S3 with PSRAM

output
20:09:43.607 -> info        Connect to new host: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_6music/bbc_6music.isml/bbc_6music-audio%3d48000.norewind.m3u8"
20:09:43.698 -> info        PSRAM found, inputBufferSize: 638965 bytes
20:09:43.698 -> [  8652][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
20:09:43.698 -> info        buffers freed, free Heap: 255448 bytes
20:09:43.744 -> info        Connection has been established in 45 ms, free Heap: 254328 bytes
20:09:43.744 -> lasthost    http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_6music/bbc_6music.isml/bbc_6music-audio%3d48000.norewind.m3u8
20:09:43.837 -> info        media sequence number: 266285342
20:09:43.837 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_6music/bbc_6music.isml/bbc_6music-audio=48000-266285342.ts"
20:09:43.872 -> [  8863][D][aac_decoder.cpp:1679] AACDecoder_AllocateBuffers(): AAC Spectral Band Replication enabled, 50788 additional bytes allocated
20:09:43.910 -> info        AACDecoder has been initialized, free Heap: 252604 bytes , free stack 5516 DWORDs
20:09:43.910 -> info        syncword found at pos 0
20:09:43.945 -> info        Channels: 2
20:09:43.945 -> info        SampleRate: 48000
20:09:43.945 -> info        BitsPerSample: 16
20:09:43.945 -> info        BitRate: 57937
20:09:43.945 -> info        AAC HeaderFormat: ADTS
20:09:43.945 -> info        AAC Codec: MPEG-4 LowComplexity
20:09:44.023 -> info        VBR recognized, audioFileDuration is estimated
20:09:44.287 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_6music/bbc_6music.isml/bbc_6music-audio=48000-266285343.ts"
20:09:48.645 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_6music/bbc_6music.isml/bbc_6music-audio=48000-266285344.ts"
20:09:55.029 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_6music/bbc_6music.isml/bbc_6music-audio=48000-266285345.ts"
20:10:02.438 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_6music/bbc_6music.isml/bbc_6music-audio=48000-266285346.ts"
20:10:08.829 -> info        Hostaddress is empty

code
```

#include "Arduino.h"
#include "WiFiMulti.h"
#include "Audio.h"

#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26
Audio audio;
WiFiMulti wifiMulti;
String ssid =     "xxxxxxx";
String password = "xxxxxx";

void setup()
   {
      Serial.begin(115200);
      WiFi.mode(WIFI_STA);
      wifiMulti.addAP(ssid.c_str(), password.c_str());
      wifiMulti.run();
      if(WiFi.status() != WL_CONNECTED){
          WiFi.disconnect(true);
          wifiMulti.run();
      }
      audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
      audio.setVolume(12); // 0...21
      audio.connecttohost("http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_6music/bbc_6music.isml/bbc_6music-audio%3d48000.norewind.m3u8"); 
  
  }

void loop()
{
    audio.loop();
   
    if(Serial.available()){ // put streamURL in serial monitor
        audio.stopSong();
        String r=Serial.readString(); r.trim();
        if(r.length()>5) audio.connecttohost(r.c_str());
        log_i("free heap=%i", ESP.getFreeHeap());
    }
 
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
```

schreibfaul1, 2024-01-02:
The BBC URLs are slightly different from the usual HLS streams. I've added a correction, the BBC streams now work.

BBC  URLs:
````
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_6music/bbc_6music.isml/bbc_6music-audio=96000.norewind.m3u8                                                         // BBC Radio 6
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio=96000.norewind.m3u8                                                // BBC Radio 1
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio=48000.norewind.m3u8                                                // BBC Radio 1
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_one_dance/bbc_radio_one_dance.isml/bbc_radio_one_dance-audio=48000.norewind.m3u8                              // BBC Radio 1 Dance
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_1xtra/bbc_1xtra.isml/bbc_1xtra-audio=48000.norewind.m3u8                                                            // BBC Radio 1xtra
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_one_relax/bbc_radio_one_relax.isml/bbc_radio_one_relax-audio=48000.norewind.m3u8                              // BBC Radio Relax
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_two/bbc_radio_two.isml/bbc_radio_two-audio=48000.norewind.m3u8                                                // BBC Radio 2
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_three/bbc_radio_three.isml/bbc_radio_three-audio=48000.norewind.m3u8                                          // BBC Radio 3
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=48000.norewind.m3u8                                       // BBC Radio 4
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_five_live/bbc_radio_five_live.isml/bbc_radio_five_live-audio=48000.norewind.m3u                               // BBC Radio 5 live
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_asian_network/bbc_asian_network.isml/bbc_asian_network-audio=48000.norewind.m3u8                                    // BBC Radio Asian Network
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_world_service/bbc_world_service.isml/bbc_world_service-audio=48000.norewind.m3u8                                    // BBC BBC World Service
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_essex/bbc_radio_essex.isml/bbc_radio_essex-audio=48000.norewind.m3u8                                          // BBC BBC Essex
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_berkshire/bbc_radio_berkshire.isml/bbc_radio_berkshire-audio=48000.norewind.m3u8                              // BBC Radio Berkshire
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_bristol/bbc_radio_bristol.isml/bbc_radio_bristol-audio=48000.norewind.m3u8                                    // BBC Radio Bristol
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_cambridge/bbc_radio_cambridge.isml/bbc_radio_cambridge-audio=48000.norewind.m3u8                              // BBC Radio Cambridge
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_coventry_warwickshire/bbc_radio_coventry_warwickshire.isml/bbc_radio_coventry_warwickshire-audio=d48000.norewind.m3u8    // BBC Radio Coventry Warwickshire
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_five_live_sports_extra/bbc_radio_five_live_sports_extra.isml/bbc_radio_five_live_sports_extra-audio=48000.norewind.m3u8  // BBC Radio 5 Live sports extra (UK Only)
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_essex/bbc_radio_essex.isml/bbc_radio_essex-audio=48000.norewind.m3u8                                          // BBC Hereford Worcester
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_berkshire/bbc_radio_berkshire.isml/bbc_radio_berkshire-audio=48000.norewind.m3u8                              // BBC - Radio Berkshire
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_bristol/bbc_radio_bristol.isml/bbc_radio_bristol-audio=48000.norewind.m3u8                                    // BBC Radio Bristo 
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_cambridge/bbc_radio_cambridge.isml/bbc_radio_cambridge-audio=48000.norewind.m3u8                              // BBC Radio Cambridge
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_cornwall/bbc_radio_cornwall.isml/bbc_radio_cornwall-audio=48000.norewind.m3u8                                 // BBC Radio Cornwall
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_cumbria/bbc_radio_cumbria.isml/bbc_radio_cumbria-audio=48000.norewind.m3u8                                    // BBC Radio Cumbria
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_cymru/bbc_radio_cymru.isml/bbc_radio_cymru-audio=48000.norewind.m3u8                                          // BBC Radio Cymru
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_derby/bbc_radio_derby.isml/bbc_radio_derby-audio=48000.norewind.m3u8                                          // BBC Radio Derby
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_devon/bbc_radio_devon.isml/bbc_radio_devon-audio=48000.norewind.m3u8                                          // BBC Radio Devon
[... cut ...]
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_wm/bbc_wm.isml/bbc_wm-audio=48000.norewind.m3u8                                                                     // BBC Radio WM
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_york/bbc_radio_york.isml/bbc_radio_york-audio=48000.norewind.m3u8                                             // BBC Radio York
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_three_counties_radio/bbc_three_counties_radio.isml/bbc_three_counties_radio-audio=48000.norewind.m3u8               // BBC Three Counties Radio

````

Alkerr, 2024-01-03:
Works perfectly. Thank you so much for the speedy resolution. Happy new year!

Alkerr, 2024-01-09:
Hi again, there's a new problem which may be related to the above. Same setup and code, seemingly only affecting BBC Radio streams. It plays fine for a while then this happens:

```
21:56:53.068 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio=96000-266380846.ts"
21:56:59.466 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio%3d96000.norewind.m3u8"
21:56:59.591 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio=96000-266380847.ts"
21:57:05.941 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio=96000-266380848.ts"
21:57:12.424 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio=96000-266380849.ts"
21:57:18.843 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio=96000-266380850.ts"
21:57:25.208 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio=96000-266380851.ts"

All fine up to here, getting a new file every 6 or 7 seconds, running solidly for 3-25 mins it seems, then:

21:57:31.687 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio%3d96000.norewind.m3u8"
21:57:31.768 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio%3d96000.norewind.m3u8"
21:57:31.849 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio%3d96000.norewind.m3u8"
21:57:31.883 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio%3d96000.norewind.m3u8"
21:57:31.966 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio%3d96000.norewind.m3u8"
21:57:31.999 -> info        new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourlw/bbc_radio_fourlw.isml/bbc_radio_fourlw-audio%3d96000.norewind.m3u8"
```

and this is happening many times a second, and it does not recover unless retuned. 
It seems to happen a lot on Radio 4 but I've also seen it on Radio 6
Any idea what it could be?

Thanks

schreibfaul1, 2024-01-10:
I can reproduce this. After 40 minutes this happened: [image] An entry in the playlist was skipped. Probably the *.m3u8 has to be fetched more often. I have set a correction, hope it helps.

Alkerr, 2024-01-10:
Running for an hour so far with no wobbles. I'd written a little code to monitor VU levels and detect when they freeze as a workaround, but this is much better! Thanks again for your help.

closed 2024-01-10 by Alkerr