schreibfaul1/ESP32-audioI2S #647: Error on compilation

renatoianhez, 2024-01-06:
I updated the library and had this error during compilation in Arduino IDE 2:

`In file included from C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioGeral\RadioeGeral\RadioeGeral.ino:6:
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:35:50: error: variable or field 'audio_id3image' declared void
 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image
                                                  ^~~~
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:35:50: error: 'File' was not declared in this scope
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:35:50: note: suggested alternative:
In file included from c:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:138,
                 from c:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:101,
                 from C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioGeral\RadioeGeral\RadioeGeral.ino:2:
C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\FS\src/FS.h:47:7: note:   'fs::File'
 class File : public Stream
       ^~~~
In file included from C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioGeral\RadioeGeral\RadioeGeral.ino:6:
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:35:56: error: 'file' was not declared in this scope
 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image
                                                        ^~~~
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:35:56: note: suggested alternative: 'pipe'
 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image
                                                        ^~~~
                                                        pipe
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:35:62: error: expected primary-expression before 'const'
 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image
                                                              ^~~~~
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:35:80: error: expected primary-expression before 'const'
 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image
                                                                                ^~~~~
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:36:51: error: variable or field 'audio_id3lyrics' declared void
 extern __attribute__((weak)) void audio_id3lyrics(File& file, const size_t pos, const size_t size); //ID3 metadata lyrics
                                                   ^~~~
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:36:51: error: 'File' was not declared in this scope
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:36:51: note: suggested alternative:
In file included from c:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:138,
                 from c:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:101,
                 from C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioGeral\RadioeGeral\RadioeGeral.ino:2:
C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\FS\src/FS.h:47:7: note:   'fs::File'
 class File : public Stream
       ^~~~
In file included from C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioGeral\RadioeGeral\RadioeGeral.ino:6:
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:36:57: error: 'file' was not declared in this scope
 extern __attribute__((weak)) void audio_id3lyrics(File& file, const size_t pos, const size_t size); //ID3 metadata lyrics
                                                         ^~~~
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:36:57: note: suggested alternative: 'pipe'
 extern __attribute__((weak)) void audio_id3lyrics(File& file, const size_t pos, const size_t size); //ID3 metadata lyrics
                                                         ^~~~
                                                         pipe
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:36:63: error: expected primary-expression before 'const'
 extern __attribute__((weak)) void audio_id3lyrics(File& file, const size_t pos, const size_t size); //ID3 metadata lyrics
                                                               ^~~~~
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:36:81: error: expected primary-expression before 'const'
 extern __attribute__((weak)) void audio_id3lyrics(File& file, const size_t pos, const size_t size); //ID3 metadata lyrics
                                                                                 ^~~~~
In file included from C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioGeral\RadioeGeral\RadioeGeral.ino:6:
c:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S\src/Audio.h:448:5: error: 'File' does not name a type; did you mean 'pipe'?
     File                  audiofile;    // @suppress("Abstract class cannot be instantiated")
     ^~~~
     pipe

exit status 1

Compilation error: exit status 1`

renatoianhez, 2024-01-16:
I encountered the same error when trying to compile for the ESP32-S3 board.
I recently reinstalled the Arduino IDE, deleting all the libraries and installing them again, one by one. Are there any prerequisites for compiling your library that I haven't noticed?

renatoianhez, 2024-01-28:
It compile (Uhay!!) after I downgrade the ESP32 in Board Manager to  2.0.11 version.

schreibfaul1, 2024-01-29:
I'm glad to read that. What happens if you upgrade the version to 2.0.14 or 3.0.0alpha3? Will the error then return?

renatoianhez, 2024-01-30:
Yes! :(  But now with other libraries... 
```
Arduino: 1.8.19 (Windows 10), Placa:"ESP32 Wrover Kit (all versions), 4MB (32Mb), Enabled, No OTA (2MB APP/2MB FATFS), 240MHz (WiFi/BT), QIO, 80MHz, 921600, None, Disabled"

In file included from C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioGeral\RadioGeral.ino:2:

C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:973:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]

  973 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!

      |        ^~~~~~~

In file included from C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32/include/driver/deprecated/driver/i2s.h:23,

                 from C:\Users\reinh\Documents\Arduino\libraries\ESP32-A2DP-main\src/BluetoothA2DPCommon.h:46,

                 from C:\Users\reinh\Documents\Arduino\libraries\ESP32-A2DP-main\src/BluetoothA2DPSink.h:18,

                 from C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioGeral\RadioGeral.ino:8:

C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]

   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"

      |  ^~~~~~~

C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32/include/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]

   27 | #warning "This set of I2S APIs has been deprecated, \

      |  ^~~~~~~

C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioGeral\RadioGeral.ino: In function 'void setup()':

C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioGeral\RadioGeral.ino:195:5: warning: missing initializer for member 'i2s_pin_config_t::mck_io_num' [-Wmissing-field-initializers]

  195 |     };

      |     ^

In file included from C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:16:

C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:973:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]

  973 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!

      |        ^~~~~~~

In file included from C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:24:

C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA(bool)':

C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: warning: missing initializer for member 'spi_bus_config_t::data4_io_num' [-Wmissing-field-initializers]

  805 |   };

      |   ^

C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: warning: missing initializer for member 'spi_bus_config_t::data5_io_num' [-Wmissing-field-initializers]

C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: warning: missing initializer for member 'spi_bus_config_t::data6_io_num' [-Wmissing-field-initializers]

C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: warning: missing initializer for member 'spi_bus_config_t::data7_io_num' [-Wmissing-field-initializers]

C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: warning: missing initializer for member 'spi_bus_config_t::isr_cpu_id' [-Wmissing-field-initializers]

C:\Users\reinh\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:829:3: warning: missing initializer for member 'spi_device_interface_config_t::clock_source' [-Wmissing-field-initializers]

  829 |   };

[... 437 lines cut ...]

      |              ~~~~~~~~~~~~^~~~~~~~~~~~~~

In file included from C:\Users\reinh\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPCommon.h:48:

C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32/include/bt/host/bluedroid/api/include/api/esp_spp_api.h:261:11: note: declared here

  261 | esp_err_t esp_spp_init(esp_spp_mode_t mode) __attribute__((deprecated("Please use esp_spp_enhanced_init")));

      |           ^~~~~~~~~~~~

Foram encontradas múltiplas bibliotecas para "SD.h"

Usado: C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\SD

Não usado: C:\Program Files (x86)\Arduino\libraries\SD

Foram encontradas múltiplas bibliotecas para "WiFi.h"

Usado: C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\WiFi

Não usado: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1
```

renatoianhez, 2024-02-02:
I completely uninstalled the Arduino IDE, now deleting folders (include in AppData) and creating a new sketches and libraries folder. Clean the registry. Reinstall all. Even with ESP32 in version 2.0.11 I got the same errors as before:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Wrover Kit (all versions), 4MB (32Mb), Enabled, No OTA (2MB APP/2MB FATFS), 240MHz (WiFi/BT), QIO, 80MHz, 921600, None, Disabled"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\reinh\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\reinh\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\reinh\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wroverkit:FlashSize=4M,PSRAM=enabled,PartitionScheme=noota_ffat,CPUFreq=240,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none,EraseFlash=none -ide-version=10819 -build-path C:\Users\reinh\AppData\Local\Temp\arduino_build_437963 -warnings=all -build-cache C:\Users\reinh\AppData\Local\Temp\arduino_cache_417401 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20221002 -prefs=runtime.tools.xtensa-esp-elf-gdb-12.1_20221002.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20221002 -prefs=runtime.tools.mklittlefs.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-12.2.0_20230208.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20221002 -prefs=runtime.tools.riscv32-esp-elf-gdb-12.1_20221002.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20221002 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-12.2.0_20230208.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.esp32-arduino-libs.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31 -prefs=runtime.tools.esp32-arduino-libs-idf-release_v5.1-3662303f31.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31 -prefs=runtime.tools.esptool_py.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.esptool_py-4.6.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.dfu-util.path=C:\Users\reinh\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\reinh\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.openocd-esp32.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230921 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20230921.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230921 -prefs=runtime.tools.esp-rv32.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.esp-rv32-2302.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-12.2.0_20230208.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.mkspiffs.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\reinh\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\reinh\Documents\4-ESP32\RadiosDados\RadioNodeMCU\RadioNodeMCU.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\reinh\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\reinh\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\reinh\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wroverkit:FlashSize=4M,PSRAM=enabled,PartitionScheme=noota_ffat,CPUFreq=240,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none,EraseFlash=none -ide-version=10819 -build-path C:\Users\reinh\AppData\Local\Temp\arduino_build_437963 -warnings=all -build-cache C:\Users\reinh\A [... cut ...]
    if (digitalRead(canalPraCima) == LOW) {//Se apertar o botao pra subir a estacao
      sobeCanal();
      mudouEstacao = true;   // Avisa la que mudou de estacao!
    }
    if (digitalRead(canalPraBaixo) == LOW) {//Se apertar o botao para descer a estacao:
      desceCanal();
      mudouEstacao = true;  //Avisa la que mudou de estacao tambem!
    }
    if (mudouEstacao) {     //Se a estacao mudou, atualiza o LCD e o terminal
      mudouEstacaoMesmo();
      mudouEstacao = false; // Mantem a estacao que esta tocando
    }
  }
}
```

renatoianhez, 2024-02-02:
Compile if I downgrade the library to version 3 (ESP 2.0.11).

Arielhh, 2024-02-24:
I encounter the same issue when the Arduino IDE version is above 2.1

closed 2024-04-09 by github-actions[bot]