schreibfaul1/ESP32-audioI2S #650: More Vorbis files that won't play

ma261065, 2024-01-08:
Hi,

I found two more Vorbis files that won't play, but do play in other players.

https://archive.org/download/gd1982-04-11.sbd.miller.108852.flac16/gd82-04-11d1t02.ogg
and
https://archive.org/download/gd1978-08-30.s2.sbd.miller.110151.flac16/gd78-08-30d3t03.ogg

Both seem to have the same problem in that there is excessive padding before the third OggS marker.

The third marker is at 0x1800, so the code in `VORBISparseOGG(uint8_t *inbuf, int *bytesLeft)` that calls `VORBIS_specialIndexOf()` doesn't find it.

I found that changing the line to:
`int idx = VORBIS_specialIndexOf(inbuf, "OggS", 8192);`
allows them to be played.

ma261065, 2024-02-01:
@schreibfaul1 what do you think about these? Is it worth adjusting the code to allow them to play?

schreibfaul1, 2024-02-02:
Oh, I think I missed your issue. I'll have a look at it soon.

schreibfaul1, 2024-02-02:
And another error fixed. Thank you for your support.

closed 2024-03-19 by github-actions[bot]