schreibfaul1/ESP32-audioI2S #652: Arduino IDE: Can't install

Vladivarius, 2024-01-09:
Both Arduino IDE 1 and 2 (both newest versions) give me the error that this is not a valid library. I tried installing .zip, unzipped folder, renamed unzipped folder. If I copy the files to Documents/Arudino/libraries/ I get invalid library error so the compiler skips over it and can't find Audio.h

schreibfaul1, 2024-01-09:
If you unpack the zip, you may have to enter a relative path, e.g. "src/audio.h".

UnexpectedMaker, 2024-01-09:
> If you unpack the zip, you may have to enter a relative path, e.g. "src/audio.h".

But why was not removed as downloadable from the library manager? 
Why was `library.properties` deleted?

schreibfaul1, 2024-01-09:
I've undone it. Why do we still need that today?

closed 2024-02-24 by github-actions[bot]