schreibfaul1/ESP32-audioI2S #656: Decoding trouble drops samples

mmar22, 2024-01-15:
I have spdif hw working now perfectly without drops on 
#define STATION "stream.funradio.sk:8000/fun128.ogg.m3u"
//#define STATION "stream.funradio.sk:8000/funpgm256.mp3.m3u"

but other stations drops in audio (issue isnt network )
//#define STATION "https://icecast.walmradio.com:8443/classic_opus"   //48k
//#define STATION "icecast2.play.cz:8000/country128aac"

can you pls check on your side what is problem , i cant test I2S on my side.
If i good seems as variable bitrates is isssue maybe

FYI i use separate task
`void task_AAloop(void)
{
  for( ;; )
  {
    audio.loop();
  //  if (!audio.isRunning())  sleep(1);
    vTaskDelay(1 / portTICK_PERIOD_MS);     //realy reqiured audio.loop fill buffer more as 1ms
    if( !audio.isRunning() && !setuponce) audio.connecttohost(STATION);  //recon
  }
}`

schreibfaul1, 2024-01-16:
Normally, an OGG frame consists of many segments with a size of a few KB to max. 64KB.
https://icecast.walmradio.com:8443/classic_opus there has only one segment, the OGG frame is <= 256 bytes. I suspect a timing problem, the ESP32-S3 manages it without stutter, the ESP32 does not. Probably the access times of the PSRAM play a role, which are shorter with the ESP32-S3.

mmar22, 2024-01-17:
Im on S3 and as i wtite used separate task . But this blocked mutex other tasks when not used vTaskDelay .(crash on wdt}
I ask exist in your lib method to setup buffering amount in one audio.loop?
Im not expert RTOS , but cant handle wiyh your loop code real multitask.

schreibfaul1, 2024-01-17:
The S3 also plays Walmradio without a task. Only with the example code on the start page.

I only used mutex in "critical sections" to avoid double accesses and collisions. Here is an example of a separate task: https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/separate_audiotask

mmar22, 2024-01-17:
Try place led blink in your example and you see 
void loop(){
    // your own code here
}

schreibfaul1, 2024-01-17:
No led, but a log every second:
````c++
void loop(){
    static uint32_t t = millis() + 1000;
    if(t < millis()){
      log_i("output every 1000ms");
      t = millis() + 1000;
    } 
    audio.loop();
}
````

mmar22, 2024-01-17:
Aghhh you simply dont understand create many tasks and your separate and test it ...
My app use 7 tasks and work ok with mp3, but not aac ogg

mmar22, 2024-01-29:
Hi i test today your latest changes and opus seems fill buffer more better. Can you pls check too AAC efectivity. I have trouble with
"icecast2.play.cz:8000/country128aac"

mmar22, 2024-01-30:
@schreibfaul1 
here my log for ok mp3
[image]

left marked number is skipped audio samples counter = need stay constant
second is xring 16k buffer fill level = more as 10k is ok

for AAC skip stay increment and buff level isnt over 8k

schreibfaul1, 2024-01-30:
This looks different in my test environment: ESP32 PSRAM deactivated,
 [image]

with PSRAM the fill level is about 70KB

mmar22, 2024-01-31:
Hmm ok i change wifi router and seems is wifi based. Now AAC works too. Thanks i close this, but on your image is 22050 samplerate, seems as somethink odd.

closed 2024-01-31 by mmar22