schreibfaul1/ESP32-audioI2S #659: audio.i2s_mclk_pin_select(I2S_MCLK); not available

ats3788, 2024-01-19:
Danke schreibfaul1 für ihre wunderbare Arbeit
In
examples/ESP32_ES8388/ESP32_ES8388.ino
is 
in line 88 audio.i2s_mclk_pin_select(I2S_MCLK);
declared but it is not in the Header.
Any clue
Martin Michael

ats3788, 2024-01-20:
Leider funktioniert es noch nicht
Seit Arduino V2 kann MCLK zusammen mit den anderen I2S Pins gesetzt werden.
Die Korrektur habe ich eingestellt, das wird jetzt funktionieren.
Muss das nun anders definiert werden.
For our English speaking friends

Unfortunately it doesn't work yet
Since Arduino V2, MCLK can be set together with the other I2S pins.
I have set the correction, it will work now.
This now needs to be defined  differently

Hast Du studiert um so tolle Projekte umzusetzen?

  ```
digitalWrite(GPIO_PA_EN, HIGH);

    // set I2S_MasterClock
    audio.i2s_mclk_pin_select(I2S_MCLK);

    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DSIN);
```

> src/main.cpp: In function 'void setup()':
> src/main.cpp:111:11: error: 'class Audio' has no member named 'i2s_mclk_pin_select'
>      audio.i2s_mclk_pin_select(I2S_MCLK);
>            ^~~~~~~~~~~~~~~~~~~
> *** [.pio\build\esp-wrover-kit\src\main.cpp.o] Error 1

schreibfaul1, 2024-01-20:
Das hab ich korrigiert, i2s_mclk_pin_select() gibts nicht mehr, [examples/ESP32_ES8388/ESP32_ES8388.ino](https://github.com/schreibfaul1/ESP32-audioI2S/commit/9140af21e8d8c3c8ac7cfeea69b34dda3d3a4a9b#diff-657fd4212fe1fe09a9be549494cd33354a5b30a1933afecd42a822995bcf0269)
[image]

closed 2024-03-05 by github-actions[bot]