schreibfaul1/ESP32-audioI2S #661: Some opus streams (low bitrate?) do not play correctly

seife, 2024-01-19:
German Deutschlandfunk opus streams do not work correctly / at all:
* http://st01.dlf.de/dlf/01/low/opus/stream.opus => opus 1ch 48000Hz, does not work at all
* http://st01.dlf.de/dlf/01/high/opus/stream.opus => opus 2ch 48000Hz, sounds strange, noticeably best with female voices.
* http://st02.dlf.de/dlf/02/low/opus/stream.opus => same with DLF Kultur
* http://st02.dlf.de/dlf/02/high/opus/stream.opus

schreibfaul1, 2024-01-19:
Hi seife,
I have only adapted the CELT decoder for the ESP32 in full band mode (48Khz). SILK is too rare to implement the SILK decoder. Especially as SILK is used for voice to save bandwidth. So the hybrid mode does not work and also no changing sample rate. I have tried to block this now.

seife, 2024-01-20:
OK, thanks for clarification :-)

closed 2024-03-05 by github-actions[bot]

clort81, 2024-11-02:
> Hi seife, I have only adapted the CELT decoder for the ESP32 in full band mode (48Khz). SILK is too rare to implement the SILK decoder. Especially as SILK is used for voice to save bandwidth. So the hybrid mode does not work and also no changing sample rate. I have tried to block this now.

Thank you so much for this awesome project!  I however do wish to inform you of a potential use case for your work, that relies particularly on SILK.

That is of a podcast player.  The bulk of available spoken-word podcasts are distributed around 64-128kbit/s.  With opus midex-mode, I compress these ~8x without exessive quality degradation.

Is there existing SILK decoder code for this CPU?  Have you looked into what full opus support might entail?

This would make this project quite unique.  Finding opus playback on handhelds is still very rare!
Cheers!