schreibfaul1/ESP32-audioI2S #669: Audio.h

Marcelverhoeven, 2024-02-01:
I have installed the https://github.com/schreibfaul1/ESP32-audioI2S.git 
But when I try to compile I get an error:
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp: In constructor 'Audio::Audio(bool, uint8_t, uint8_t)':
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:206:18: error: 'struct i2s_config_t' has no member named 'mclk_multiple'
     m_i2s_config.mclk_multiple        = I2S_MCLK_MULTIPLE_128;
                  ^
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:206:41: error: 'I2S_MCLK_MULTIPLE_128' was not declared in this scope
     m_i2s_config.mclk_multiple        = I2S_MCLK_MULTIPLE_128;
                                         ^
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:212:65: error: 'I2S_COMM_FORMAT_STAND_MSB' was not declared in this scope
         m_i2s_config.communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_MSB); // vers >= 2.0.5
                                                                 ^
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:222:65: error: 'I2S_COMM_FORMAT_STAND_I2S' was not declared in this scope
         m_i2s_config.communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_I2S); // Arduino vers. > 2.0.0
                                                                 ^
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp: In member function 'const char* Audio::parsePlaylist_M3U8()':
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:2563:45: error: 'lltoa' was not declared in this scope
                     lltoa(xMedSeq, llasc, 10);
                                             ^
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp: In member function 'uint64_t Audio::m3u8_findMediaSeqInURL()':
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:2786:31: error: 'lltoa' was not declared in this scope
             lltoa(b, llasc, 10);
                               ^
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp: In member function 'bool Audio::setPinout(uint8_t, uint8_t, uint8_t, int8_t)':
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4508:18: error: 'struct i2s_pin_config_t' has no member named 'mck_io_num'
     m_pin_config.mck_io_num = MCLK;
                  ^
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp: In member function 'void Audio::setI2SCommFMT_LSB(bool)':
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4686:65: error: 'I2S_COMM_FORMAT_STAND_MSB' was not declared in this scope
         m_i2s_config.communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_MSB); // v >= 2.0.0
                                                                 ^
c:\Users\marce\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4690:65: error: 'I2S_COMM_FORMAT_STAND_I2S' was not declared in this scope
         m_i2s_config.communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_I2S); // vers >= 2.0.0
                                                                 ^

exit status 1

Compilation error: exit status 1

===> Please advise

closed 2024-03-18 by github-actions[bot]

kalodj, 2024-04-13:
I have exactly same problem :( .. does anybody know what i did wrong ? Is there a solution on it .. all help much appreciated ! Thx

whatever sketch i compile i get this error, e.g. on the example on main page (https://github.com/schreibfaul1/ESP32-audioI2S):

====
C:\Users\Admin\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp: In constructor 'Audio::Audio(bool, uint8_t, uint8_t)':

C:\Users\Admin\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:207:18: error: 'struct i2s_config_t' has no member named 'mclk_multiple'
     m_i2s_config.mclk_multiple        = I2S_MCLK_MULTIPLE_128;
                  ^
C:\Users\Admin\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:207:41: error: 'I2S_MCLK_MULTIPLE_128' was not declared in this scope
     m_i2s_config.mclk_multiple        = I2S_MCLK_MULTIPLE_128;
                                         ^
...

I am using the ESP32 Dev (38 pins): https://www.az-delivery.de/en/products/esp32-developmentboard

kalodj, 2024-04-13:
Screenshots:
[image]
[image]
[image]

kalodj, 2024-04-13:
found your solution here: .. https://github.com/techiesms/ESP32-ChatGPT-V2/issues/1
THANK YOU !