schreibfaul1/ESP32-audioI2S #671: Is it possible to play an mp3 from char array?

Emmanuele75, 2024-02-04:
Hi everyone!
I would like to play a sound (beep) by loading it directly into a char array.
Example:

const unsigned char beep_mp3[] = {
   0x49, 0x44, 0x33, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x19, 0x54, 0x53,
   0x53, 0x45, 0x00, 0x00, 0x00, 0x0f, 0x00, 0x00, 0x03, 0x4c, 0x61, 0x76,
.........

There's no way I can get this.
Can anyone help me?
Thank you.

akakhtar, 2024-02-12:
Use tone function of arduino to play beep sound

sartyx, 2024-03-01:
Same question here and no, it is not about beeps. I just want to store the audio internally because I don't want to add any external memory such as SD or even a stream.

Doesn't have to be mp3 for my usecase though.

```
unsigned const char WavData[35564] = {
  0x52, 0x49, 0x46, 0x46, 0xe4, 0x8a, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45,
  0x66, 0x6d, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x01, 0x00, 
[...] and so on 
```

amarotica, 2024-03-29:
I'm looking for the same functionality. I converted a WAV to RAW to a header file and a feature that could then play that included .h file would be awesome!

schreibfaul1, 2024-03-29:
Why should it be so complicated? You can save the audio file directly as SPIFFS or in an FFat partition without RAW conversion.

sartyx, 2024-03-29:
There are reasons for that, first is, not all ESP32 support SPIFF even if they should second is data from an array is faster which can be an issue if you want to play sounds on button press for a game.

robertmuth, 2024-04-06:
A design change the would make this easier: 
* introduce an AudioStream abstraction (maybe the regular Stream that File inherits from is suitable).
* replace the member variable "File              audiofile" with  "AudioStream* m_stream" or similar (maybe a unique_ptr)
* provide a way to set m_stream with a public API

With this change, if I have an in-memory file, I could just write a simple AudioStream wrapper for it and hand it to class Audio for playing.

Maybe Audiostream can be made general enough to also include web streams which would allow for moving a lot of code out
of the class Audio.

closed 2024-05-22 by github-actions[bot]