schreibfaul1/ESP32-audioI2S #674: Problem with "D0WD Rev 1" chip and PSRAM

Pako2, 2024-02-10:
I am using this excellent ESP32audio-I2S library in my project. I rotate several test HW. I ran into an interesting problem.
In some cases, lines like these appear in the serial monitor:
```
10:27:02.093 > I2S AUDIO_INFO        AAC decode error -15 : NCHANS_TOO_HIGH
10:27:02.100 > I2S AUDIO_INFO        syncword found at pos 511
10:27:02.104 > I2S AUDIO_INFO        syncword found at pos 0
10:27:02.108 > I2S AUDIO_INFO        AAC decode error -3 : INVALID_ADTS_HEADER
10:27:02.113 > I2S AUDIO_INFO        syncword found at pos 7
10:27:02.118 > I2S AUDIO_INFO        syncword found at pos 0
10:27:02.122 > I2S AUDIO_INFO        AAC decode error -3 : INVALID_ADTS_HEADER
10:27:02.126 > I2S AUDIO_INFO        syncword found at pos 11
10:27:02.131 > I2S AUDIO_INFO        syncword found at pos 0
10:27:02.135 > I2S AUDIO_INFO        AAC decode error -3 : INVALID_ADTS_HEADER
10:27:02.140 > I2S AUDIO_INFO        syncword found at pos 6
10:27:02.144 > I2S AUDIO_INFO        syncword found at pos 0
10:27:02.151 > I2S AUDIO_INFO        Channels: 2
10:27:02.155 > I2S AUDIO_INFO        SampleRate: 22050
10:27:02.159 > I2S AUDIO_INFO        BitsPerSample: 16
10:27:02.162 > I2S AUDIO_INFO        BitRate: 92334
10:27:02.166 > I2S AUDIO_INFO        AAC HeaderFormat: ADTS
10:27:02.170 > I2S AUDIO_INFO        AAC Codec: MPEG-4 LowComplexity
```
It is usually accompanied by an unpleasant sound effect.
Of course, I investigated what the occurrence of the mentioned phenomenon depends on.
After many hours I came to a very interesting result:
This issue occurs only when all of the following conditions are true:
1. The chip model is ESP32-D0WD Rev 1
2. PSRAM is enabled
3. The program is compiled in the VS Code/PlatformIO environment
Just change one of the listed points and the problem does not occur (or did I not wait long enough?).

This means that:
1. ESP32-D0WD-V3 Rev 3 chips do not suffer from this problem
2. If PSRAM is not enabled (or installed), the problem does not occur
3. If I compiled the program using the Arduino IDE, the problem never occurred

Note:
If I'm using PSRAM with a D0WD Rev 1 chip, I also use the build_flag "-mfix-esp32-psram-cache-issue".

This is the program I tested with:
```
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"

#include "esp_log.h"

#define I2S_DOUT     5
#define I2S_BCLK    19
#define I2S_LRC     18

const char *TAG = "main"; // For debug lines
Audio audio;
String ssid =     "XXXXX";
String password = "XXXXX";

void setup() {
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolumeSteps(100);
  audio.setVolume(40);
  vTaskDelay(4000 / portTICK_PERIOD_MS);          // Wait for serial monitor to start
  Serial.begin(115200);
  ESP_LOGW ( TAG, "Starting ...") ;
  ESP_LOGW ( TAG, "ESP32 Chip model:  %s Rev %d", ESP.getChipModel(), ESP.getChipRevision());
  ESP_LOGW ( TAG, "SdkVersion:        %s",                            ESP.getSdkVersion());
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid.c_str(), password.c_str());
  while (WiFi.status() != WL_CONNECTED)
  {}
  vTaskDelay(4000 / portTICK_PERIOD_MS);
  audio.connecttohost("http://rozhlas.stream/dvojka_high.aac");
}

void loop()
{
  audio.loop();
}

void audio_info(const char *info){
    Serial.print("I2S AUDIO_INFO        "); Serial.println(info);
}
```
The problem is, that:
1. I have several modules with the D0WD Rev 1 chip and I would like to use them
2. My project is quite large and without PSRAM I soon run out of memory
3. Some libraries I use need some patches. This is very inconvenient when using the Arduino IDE. I need PlatformIO.

schreibfaul1, 2024-02-10:
If PSRAM is available, the decoders will use it. I think there are other workarounds besides mfix-esp32-psram-cache-issue. If you use PlatformIO, then use Arduino as a component. Then you have access to the PSRAM settings via menuconfig.

Pako2, 2024-02-11:
I thank you for the answer. Of course, I was already looking for possible solutions. Unfortunately, I didn't find anything useful. That's why I reported it here - with the hope that there is already some concrete solution.
I have zero experience with settings using menuconfig. I won't go into it because I expect it would require many hours of testing with an uncertain result.
So I prefer to avoid ESP32-D0WD Rev 1 chips.
Hopefully my post will at least help other users.
If they encounter the same problem, they will no longer have to search for which combination of input conditions is unsuitable for use with the ESP32audio-I2S library.

mmar22, 2024-02-11:
Maybe good is add info about versions. For example test you more versions or only
framework-arduinoespressif32@3.20011.230801 ...
or only latest 3.20014.231204

Pako2, 2024-02-12:
I only use the latest version: framework-arduinoespressif32 @ 3.20014.231204 (2.0.14). This is the same for both Arduino IDE and PlatformIO.

closed 2024-03-28 by github-actions[bot]