schreibfaul1/ESP32-audioI2S #679: WiFiClientSecure fails just by having the Audio object created

codegrue, 2024-02-13:
Doing nothing more than having this line:

`Audio audio;`

Causes all SSL API calls to fail with either memory issue or certificate verification failure:

`[start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed`

I suspected the heap getting too small, but I still have 40000 bytes. Are there any known interactions between ESP32-audioI2S and WifiClientSecure?

codegrue, 2024-03-16:
I did confirm that this was a memory related issue. I was unable to get Audio to use PSRAM, and WifiClientSecure also was using main heap. I did figure out how to solve the second item, but I still suspect Audio won't use it because implementing it as a global variable, it get's created before the ESP32 PSRAM can be initialized in app_main(void). I did end up moving to a different audio stack before I was able to solve it, so I'm not sure if the technique would work where you create Audio as a static in a function (on first use essentially) rather than as a global.

closed 2024-03-16 by codegrue

kungfupizza, 2024-06-18:
@codegrue Can I ask how were you able to move audio task as a static function?
Also, what other audio stack did you move to?

codegrue, 2024-06-18:
I was not able to get the Audio to work. The library I switched to is:

https://github.com/pschatzmann/arduino-audio-tools