schreibfaul1/ESP32-audioI2S #681: Trouble playing "https://s4.yesstreaming.net:17199/stream"

FatherMarco1971, 2024-02-16:
I'm experimenting the library to understand and learn and because can be what i need for a web radio project.

Anyway i'm testing in two scenario:

1. esp32 no psram + PCM5102A
2. ai-thinker 2.2 esp32-A1s 4mb psram es8288

Playing the stream [https://s4.yesstreaming.net:17199/stream](url) on both platform result in chopping audio e continuos slow stream,.... warning (and I'm sure is not absolutely possible my network is slow :-)) to prove this the stream sound perfectly with vlc and on both platform using different library (arduino-audio-tool+ arduino-audio-kit).
Maybe the management of the stream can be optimized?

FatherMarco1971, 2024-02-19:
ok I see your point...We have to recompile wifi lib to enhance WiFi performance...
I'll try to see to learn something new but onestly i'm not sure is the best way since with other lib everithing could work...but i'm not sufficently expert to say something.
Anyway on the paper your lib is the only one could play m3u8 so i've to learn how to recompile wifi lib :-)

FatherMarco1971, 2024-02-20:
Just to close this: recompiling with platformio following your suggestion regarding enhance WiFi work like a charme!

dabu667, 2024-02-27:
@FatherMarco1971 could you help me with some general guidelines on how to improve the streaming reception?
How did you manage to get rid of the "slow stream.." message and audio dropouts?