schreibfaul1/ESP32-audioI2S #685: M3U8 noisy

FatherMarco1971, 2024-02-20:
Using releas 3.0.8 with this stream

http://streamcdnm2-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8
Cuase the stop after the (perfect) reproduction of 3 ts.
Using the 3.0.8p it doesn't stop but the sound is orrible aand full of interruption and glitch and the log is full of:

AAC decode error -15 : NCHANS_TOO_HIGH

I'm not good enough to debug your code but if you need help for testinf i'll do my best

schreibfaul1, 2024-02-20:
There is other information in the stream contained. Could be a video sequence, I don't know. I copied two streams from the website, [from the station](https://www.deejay.it/webradio/), which are pure audio https://streamcdnf25-4c4b867c89244861ac216426883d1ad0.msvdn.net/webradio/deejay80/live.m3u8 https://streamcdnf24-4c4b867c89244861ac216426883d1ad0.msvdn.net/webradio/deejayonelove/live.m3u8

FatherMarco1971, 2024-02-21:
I understand.
Thank you very mutch.

FatherMarco1971, 2024-03-03:
I think the problem is not the presence of any video but that the AAC decoder can't process correctly audio.
Fromm VLC i can see for this [http://streamcdnm2-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8](url) stream:
[image]

for this [https://streamcdnf24-4c4b867c89244861ac216426883d1ad0.msvdn.net/webradio/deejayonelove/live.m3u8](url):
[image]

I think the problem is that "extensio AAC: SBR".
The courios thing is that  the version 3.0.8 sound perfectly but stop after 3 .ts.

schreibfaul1, 2024-03-04:
Here is a recording from the link above als RAW file
[raw.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/14484638/raw.txt)
According to the specification, a ts packet is exactly 188 bytes long. This is not the case here. I don't know what is transmitted between the packets. Maybe someone who reads here can help.
[image]

FatherMarco1971, 2024-03-04:
Rolling back to this release of your lib (10 Jan 2024):

[https://github.com/schreibfaul1/ESP32-audioI2S/tree/17f0e0468ee862b9ea1d4b6b85d91b599535acdc](url)

Everething work like a charme!!
So I was wrong your code can work but some chenges done from 10 Jan to today have break something!!!!!

I try my best but your code is too complex for my capability to understand where the problem is but, if you need log or testing i'm happy to do my best.

FatherMarco1971, 2024-03-04:
I've found what cause the bug @schreibfaul1
I will create a pull request so you can review it and accept or modify it.

closed 2024-03-07 by FatherMarco1971