schreibfaul1/ESP32-audioI2S #701: setTimeOffset doesn't work as expected

ewaca, 2024-03-17:
Hi,

I want to use setTimeOffset to fast forward in mp3 file, with 10 sec steps, but the mp3 file position is not always increased by 10.
in the loop function I print the audioCurrentTime every 0.2 seconds
When i press the right button, i set a 10 secondes time offset 
Here is what i get in serial terminal

```
As the Pages Burn.mp3
0
0
0
0
0
1
1
1
1
1
2
2
2
2
2
3
3
3
3
3
4
4
Current time : 4
set time offset : 10 Right
8
8
8
8
8
8
9
9
9
9
9
10
10
10
10
10
10
11
11
11
11
11
Current time : 11
set time offset : 10 Right
24
24
24
24
24
24
25
25
25
25
25
26
26
26
26
26
27
27
27
27
Current time : 27
set time offset : 10 Right
40
40
40
40
40
40
41
41
41
41
41
42
42
42
Current time : 42
set time offset : 10 Right
48
48
48
48
48
48
49
49
49
49
49
50
50
50
50
50
Current time : 50
set time offset : 10 Right
64
64
64
64
64
64
65
65
65
65
65
66
66
66
66
66
Current time : 66
set time offset : 10 Right
80
80
80
80
80
81
81
81
81
81
81
82
Current time : 82
set time offset : 10 Right
88
88
88
88
88
88
89
89
89
89
89
90
90
90
90
90
90
91
91
91
91
91
92
92
Current time : 92
set time offset : 10 Right
104
104
104
104
104
104
105
105
105
105
```

First offset : 4->8 = 4 sec
2nd offset 11->24 = 13 sec
3th 27->40 = 13 sec
4th 42->48 = 6 sec
5th 50 ->64 = 14 sec
6th : 56->80 => 14 sec
7th : 82->88 => 6 sec
8th : 92->104 => 12 sec

here my code for the right button press  
`  if(Status=="Playing"){
      act=audio.getAudioCurrentTime();
      Serial.print("Current time : ");
      Serial.println(act);
      Serial.print("set time offset : 10 ");
      audio.setTimeOffset(10);`

schreibfaul1, 2024-03-17:
For a audio file, I know the length of the audio block after the metadata and embedded images have been removed. Unfortunately, I don't know anything about the total playing time. If the file runs for a few mp3 frames, I can calculate the bitrate (or rather estimate it in the case of VBR). This is a problem for which I have no solution. If anyone reading here has a good idea of how to do this better with the resources offered by the ESP32, please present your idea here.

ewaca, 2024-03-17:
there is a opensource project for MP3 audio player that seems to use its own MP3 decoder library. I have not find fast forward function yet in the code, but may be you can find any ideas on how to deal with in their code : https://codeberg.org/cool-tech-zone/tangara-fw

closed 2024-05-01 by github-actions[bot]