schreibfaul1/ESP32-audioI2S #702: error when switching from 3.08 to latest version

Arielhh, 2024-03-19:
Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:35:50: error: variable or field 'audio_id3image' declared void
 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image
it looks like there is a problem with the declaration of the function audio_id3image in the Audio.h file of the ESP32-audioI2S library. The error specifically states that the variable or field audio_id3image is declared as void, which is likely not what was intended.

schreibfaul1, 2024-03-19:
Hi Arielhh, 
that would mean that in line 35 in Audio.h 
`external __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image`
 position 50 **class fs::File** is causing a problem.
Unfortunately I cannot reproduce the error, is this related to Arduino V3.0.0?

Arielhh, 2024-03-19:
I am using the latest Arduino release 2.3.2 . My code is long so i cannot post it . it compiles with no problem when using ESP32-audioI2S - V3.08.

Arielhh, 2024-03-20:
Hi Wolle, I looked at the two versions of Audio.h files. In the old version there are additional included libraries. When I am adding these libraries to the new version the code is compiling correctly. were this libraries omitted deliberately? In my code i am using SPIFFS and i believe that the missing library from the new version is causing the compilation failure.
new version: #pragma once
#pragma GCC optimize ("Ofast")
#include <vector>
#include <Arduino.h>
#include <libb64/cencode.h>
#include <esp32-hal-log.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <FS.h>
#include <atomic>

Old version:
#pragma once
#pragma GCC optimize ("Ofast")
#include <vector>
#include <Arduino.h>
#include <libb64/cencode.h>
#include <esp32-hal-log.h>
#include <SPI.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <SD.h>
#include <SD_MMC.h>
#include <SPIFFS.h>
#include <FS.h>
#include <FFat.h>
#include <atomic>

schreibfaul1, 2024-03-20:
Hi Arielhh, thanks for narrowing down the error. I have added the libraries again.

Arielhh, 2024-03-20:
Wolle, thanks for developing this great library :)

closed 2024-03-20 by Arielhh