schreibfaul1/ESP32-audioI2S #703: Cannot get values from GPIO after setPinout

Turboscherbe, 2024-03-19:
Thanks for this great library. However I found an issue I was not able to resolve for days:

I am using an ESP32 D1 mini with components connected to pins as follows:

```c
// MAX98357A
#define I2S_DOUT 25
#define I2S_BCLK 26
#define I2S_LRC  27
// SD card
#define SPI_SCK  18
#define SPI_MISO 19
#define SPI_MOSI 23
#define SD_CS    5
// DCF 77
#define DCF77  13
```
I'm using the sample code from the README.md to play wav files from SD card which is working well. As you can see, I also have a DCF77 receiver connected to GPIO13 (also tried 16,17) for simply reading values ```pinMode(DCF77, INPUT);  val = digitalRead(DCF77); ``` in the main loop which is also working well.

But putting both together (audio playing and reading DCF77 value), the ```digitalRead(DCF77)``` is always returning 0 and no data anymore. After trial and error I found out, that it is starts returning 0-values after calling ```audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);```. Does setPinout overwrite other GPIOs? Do I need to configure additional GPIOs in a certain way to use them together with your audio lib (i.e. interrupts...)?

Thanks for any help

Turboscherbe, 2024-03-19:
Here is the minimal sketch to reproduce:

```c++
#include "Audio.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
#define I2S_DOUT    25
#define I2S_BCLK    26
#define I2S_LRC     27
#define SPI_SCK     18
#define SPI_MISO    19
#define SPI_MOSI    23
#define SPI_SS      5
#define DCF77       16

Audio audio;

void setup() {
    pinMode(DCF77, INPUT);
    pinMode(SPI_SS, OUTPUT);      
    digitalWrite(SPI_SS, HIGH);

    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    Serial.begin(115200);
    SD.begin(SPI_SS);
    
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(1);
    audio.connecttoFS(SD, "/kunst.wav");
}

void loop()
{
    audio.loop();

    // Output DCF77 value
    int i = digitalRead(DCF77);               // <--- i is always 0!!!! when commenting 
                                              //      out  `audio.setPinout(...)` above, 
                                              //      then this returns correct values. Of course
                                              //      the sound output then will not work
    Serial.println((String)"DCF77: " + i);
}
```

schreibfaul1, 2024-03-19:
Interference between the GPIOs is not very likely. The clocks at the I2S output are close to the frequency of the DCF77. I suspect that nothing switches at the DCF77 output if there is a signal at I2S. Only good shielding between the digital part and the DCF77 receiver will help.

Turboscherbe, 2024-03-19:
Wow, thank you so much! After finding some alu-foil from the kitchen and wrapping the DCF77 into it - I was able to read the digital input again. I would never have found that out without your advice, thanks again :-)

closed 2024-03-19 by Turboscherbe