schreibfaul1/ESP32-audioI2S #710: PSRAM leakage?

SanZamoyski, 2024-03-28:
Hi!

I think that something is not ok here. When song ends or is rewinded to next my free PSRAM lowers.
I've left my player overnight and result is recorded in attached txt file.

[[24.03.19]-mod.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/14789437/24.03.19.-mod.txt)

I can not provide at the moment right software version, since it was modified since then. But I'll try to give You some useful  informations. I'm using platformio:

[platformio.ini.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/14789484/platformio.ini.txt)
[UAP_PinDef.inch.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/14789493/UAP_PinDef.inch.txt)
[UAP_SdArrays.inc.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/14789500/UAP_SdArrays.inc.txt)
[UAP_DBG.inc.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/14789510/UAP_DBG.inc.txt)
[main.cpp.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/14789522/main.cpp.txt)
[UAP_CoreLoop.inc.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/14789532/UAP_CoreLoop.inc.txt)

Can we delete those files after solving/nofix this problem?

Is there anything wrong in my source or this is a bug? Sorry for the mess in source.

Best regards!

All my audio files are full-albums flac.

schreibfaul1, 2024-03-28:
The memory leak will have existed for a long time, the samplesBuffer was not released correctly. Thanks for the tip.

SanZamoyski, 2024-03-28:
Can You provide some workaround?
I.e. what should I do if PSRAM gets too low to play another song?

i```
f(ESP.getFreePsram() < 32*1024){
  //do something smart here...
}
```

Currently I'm rebooting ESP when I realise there is PSRAM shortage.

schreibfaul1, 2024-03-28:
The loss was about 64KB at each initialisation of the FLAC decoder. I used `ESP.getFreePsram()` too  to plug the leak.

SanZamoyski, 2024-03-28:
but how?

FatherMarco1971, 2024-03-31:
@SanZamoyski  he has committed a correction or i'm not understandign what you are asking for?

[image]

PMKrol, 2024-04-03:
I have checked it. Still free PSRAM lowers each time I play next flac.

SanZamoyski, 2024-05-06:
Updated libraries, it is in fact working now. Thanks!

closed 2024-05-06 by SanZamoyski