schreibfaul1/ESP32-audioI2S #715: Audio clipping when played off of an SD card.

DJX10, 2024-04-07:
Hi!

I'm currently trying to use the project for a ESP32 powered door bell chime, but I'm running into an issue where when I try to play audio via an SD card, the audio clips. I'm using an ESP32-WROOM-32 with a MAX98357A amplifier. Whats odd here is that if I use the MP3 stream in the example, it plays back just fine, which to me rules out the amplifier as being the issue. Here is my project code: 
`#define BLYNK_TEMPLATE_ID "**************"
#define BLYNK_TEMPLATE_NAME "*************"
#define BLYNK_FIRMWARE_VERSION "0.1.2"

#define BLYNK_PRINT Serial
#define APP_DEBUG
#define USE_ESP32_DEV_MODULE

#include "BlynkEdgent.h"
#include <Arduino.h>
#include "Audio.h"
#include "SPI.h"

#define SD_CS 5
#define SPI_MOSI 23
#define SPI_MISO 19
#define SPI_SCK 18
#define I2S_DOUT 25
#define I2S_BCLK 27
#define I2S_LRC 26

#define LED_PIN 2  // Use pin 2 for LED (change it, if your board uses another pin)

Audio audio;

void setup()
{
    

    pinMode(SD_CS, OUTPUT);
    digitalWrite(SD_CS, HIGH);
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    SPI.setFrequency(1000000);

    Serial.begin(115200);
    SD.begin(SD_CS);
    delay(100);

    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21);

    audio.connecttoFS(SD, "1.mp3");

    BlynkEdgent.begin();
}

void loop()
{
   BlynkEdgent.run();
   audio.loop();
}

BLYNK_WRITE(V0) {

   int value = param.asInt();

  if (value == 1) {
    digitalWrite(LED_PIN, HIGH);
    Serial.print("value =");
    Serial.println(value);
    audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.mp3");
  } else {
    digitalWrite(LED_PIN, LOW);
    Serial.print("value = ");
    Serial.println(value);
  }
}`
Here the project plays the test chime file "1.mp3" on boot and I can trigger the the mp3 stream via a Blynk routine.

The test audio file I'm using is just a simple chime I made using Logic Pro and then resampled to a 128 kbps MP3 file in Audacity. I've attached it as well as what is heard from the speaker to this post (Git wouldn't let me upload audio files, had to put both files into a ZIP. Any help is greatly appreciated. Thank you!
[Archive.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/14895453/Archive.zip)

schreibfaul1, 2024-04-08:
Sounds like "BlynkEdgent.run()" needs a lot of processing time. If this is the case, audio and BlynkEdgent should run on different cores

DJX10, 2024-04-08:
Good idea, I gave that shot by disabling all the BlynkEdgent code and just had the audio code running, unfortunatly the issue remained, without the Blynk code running. One other thing I noticed is when I tried WAV files, the frequency of the "clipping" or "pulsing" would change, If I tried 8 bit WAV files, the amplitude would pulse roughly at the same rate as the mp3 files, if I used 16 bit, it would pulse at a much faster rate. Let me know if you want to hear what the WAV files sound like and I can upload samples. It's still bizzare to me that it has no issue with the mp3 stream, just when I play any audio file from the SD card.

schreibfaul1, 2024-04-10:
This is a question of data throughput, i.e. the sample rate and the compression ratio. WAV is uncompressed. Audio.loop() should be called 50 times or more per second.

DJX10, 2024-04-26:
Sorry for the late reply, had to step away from this project for work related issues, it does appear to be throughput related, when nothing is playing, the loop is running 1000 times per second, however, whenever I play audio from the SD card, the loop rate drops to 39 loops per second and holds steady. Note, this is with only the program code running that plays audio, no other code was running in the loop. How can I get that loop rate up to meet that minimum of 50 cycles? 

Thank you!

closed 2024-06-09 by github-actions[bot]