schreibfaul1/ESP32-audioI2S #716: I2S data input as source

AcuarioCat, 2024-04-10:
I´m using an RD5807 as one possible source for audio. I´m using the I2S output from the RD5807 and processing it to control the volume before sending it to I2S output (to a MAX98357).

I also have SD and Web stream sources for input to Audio, these work correctly.

Is there a way to use I2S as a source? 
The problem I have is switching from SD to the RDS, I end up with a garbled output that is of no use.

Switching from the RDS to SD is fine.
If I reset my code with RDS selected it´s fine until I switch to SD then back to RDS.

`
//Handle reading i2s, changing the volume and sending it to the i2s amplifier 
void rds_audio_task(void* arg) {
    static int16_t raw_samples[SAMPLE_BUFFER_SIZE];
    size_t bytes_written;

    while (1) {
        // read from the I2S device
        size_t bytes_read = 0;
        i2s_read(I2S_NUM_1, raw_samples, sizeof(int16_t) * SAMPLE_BUFFER_SIZE, &bytes_read, portMAX_DELAY);
        int samples_read = bytes_read / sizeof(int16_t);

        for (int i = 0; i < samples_read; i++)
        {
            raw_samples[i] = raw_samples[i] * RDSRadio.radioVolume;
        }

        i2s_write(I2S_NUM_0, raw_samples, bytes_read, &bytes_written, portMAX_DELAY);
        if (!RDSRadio.radioActive)
        {
            hzRadioTask = NULL;
            vTaskDelete(NULL);
        }
        vTaskDelay(1);
    }
}
`

schreibfaul1, 2024-04-10:
I'm not sure if I understood your question correctly, RD5807 is an FM tuner with analogue output. You could then use a PCM5102A as DAC and switch the outputs from the DAC and the tuner to a common amplifier. I don't think the MAX98357 has an analogue input to do this.

AcuarioCat, 2024-04-10:
The RDA5807FP can be configured for analogue output or I2S output by configuring register 04. This is only available on the FP device.

What I have found is that there is no control over volume on the I2S output. The volume only works on the analogue output.
My solution is to connect the I2S output the second ESP32 I2S channel and then control the volume by processing the sampled data.

So I then need to send the modified data to theESP32 I2S output, and this is the reason for my question.
Can I use I2S data as an input source to the library?

closed 2024-05-25 by github-actions[bot]