schreibfaul1/ESP32-audioI2S #719: How to reduce library size?

AnatolyNevzoroff, 2024-04-11:
Great library!
But very hard! The library takes up about 90% of the memory. I have never used and do not intend to use opus_decoder.h and vorbis_decoder.h.
These decoders take up memory. How to exclude them from the library. If you use ESP32 only for radio, you can exclude flac_decoder.h, but how to do this?
I know a project that uses your library without the specified decoders. The files: audio.cpp and audio.h are modified, but I can’t understand how this was done.
Help me please!

schreibfaul1, 2024-04-12:
You probably mean the flash memory. Please try another partition table, such as "Huge App (3MB no OTA/ 1MB SPIFFS)"

AnatolyNevzoroff, 2024-04-16:
Yes, I mean memory for recording a sketch.
I understand that you will not exclude decoders just because “someone asked,” but it seems to me that sooner or later, such a need will still arise. Think about this question.
The option you suggested does not suit me...

closed 2024-05-30 by github-actions[bot]

whitevast, 2025-12-23:
The question is relevant. Why keep unnecessary decoders in memory to the detriment of OTA?