schreibfaul1/ESP32-audioI2S #720: Help wtih https://streamcdnc2-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/WjpMtPyNjHwj/chunklist_b128000.m3u8

FatherMarco1971, 2024-04-12:
I don't understand why it is not played...
It seems the m3u8 is parse wrongly?

I think the error is here:

```
https://streamcdnc2-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/WjpMtPyNjHwj/#MY-USER-CHUNK-DATA-1:ON-TEXT-DATA="20240412140806"

```
It should be something like:

```
https://streamcdnc2-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/WjpMtPyNjHwj/media-u4e34f9bu_b128000_195542512.aac

```
but i don't understand how to correct it....

```6:08:27.622 > [0;32m[  3882][I][Audio.cpp:5375] ts_parsePacket(): parseTS reset
16:08:30.097 > [0;32m[  6357][I][Audio.cpp:3888] parseHttpResponseHeader(): now parse playlist[0m
16:08:30.105 > [  6360][I][Audio.cpp:2518] readPlayListData(): pl=0 "#EXTM3U"
16:08:30.112 > [  6360][I][Audio.cpp:2518] readPlayListData(): pl=1 "#EXT-X-VERSION:3"m
16:08:30.119 > [  6367][I][Audio.cpp:2518] readPlayListData(): pl=2 "#EXT-X-TARGETDURATION:4"
16:08:30.127 > [  6375][I][Audio.cpp:2518] readPlayListData(): pl=3 "#EXT-X-MEDIA-SEQUENCE:195542511"
16:08:30.135 > ;32m[  6384][I][Audio.cpp:2518] readPlayListData(): pl=4 "#EXT-X-DISCONTINUITY-SEQUENCE:0"0m
16:08:30.145 > [  6392][I][Audio.cpp:2518] readPlayListData(): pl=5 "#EXTINF:2.944,"[0m
16:08:30.151 > m[  6399][I][Audio.cpp:2518] readPlayListData(): pl=6 "#MY-USER-CHUNK-DATA-1:ON-TEXT-DATA="20240412140806""
16:08:30.161 > ;32m[  6410][I][Audio.cpp:2518] readPlayListData(): pl=7 "media-u4e34f9bu_b128000_195542511.aac"
16:08:30.170 > [0;32m[  6419][I][Audio.cpp:2518] readPlayListData(): pl=8 "#EXTINF:3.008,"
16:08:30.178 > [0;32m[  6426][I][Audio.cpp:2518] readPlayListData(): pl=9 "#MY-USER-CHUNK-DATA-1:ON-TEXT-DATA="20240412140809""0m
16:08:30.188 > [  6436][I][Audio.cpp:2518] readPlayListData(): pl=10 "media-u4e34f9bu_b128000_195542512.aac"[0m
16:08:30.197 > 2m[  6445][I][Audio.cpp:2518] readPlayListData(): pl=11 "#EXTINF:3.008,"
16:08:30.204 > ;32m[  6453][I][Audio.cpp:2518] readPlayListData(): pl=12 "#MY-USER-CHUNK-DATA-1:ON-TEXT-DATA="20240412140812""
16:08:30.215 > [  6463][I][Audio.cpp:2518] readPlayListData(): pl=13 "media-u4e34f9bu_b128000_195542513.aac"0m
16:08:30.224 > [  6472][I][Audio.cpp:2518] readPlayListData(): pl=14 "#EXTINF:3.008,"[0m
16:08:30.231 > 2m[  6479][I][Audio.cpp:2518] readPlayListData(): pl=15 "#MY-USER-CHUNK-DATA-1:ON-TEXT-DATA="20240412140815""
16:08:30.242 > [0;32m[  6490][I][Audio.cpp:2518] readPlayListData(): pl=16 "media-u4e34f9bu_b128000_195542514.aac"
16:08:30.251 > [  6499][I][Audio.cpp:2518] readPlayListData(): pl=17 "#EXTINF:3.008,"[0m
16:08:30.258 > [  6506][I][Audio.cpp:2518] readPlayListData(): pl=18 "#MY-USER-CHUNK-DATA-1:ON-TEXT-DATA="20240412140818""
16:08:30.268 > 0;32m[  6517][I][Audio.cpp:2518] readPlayListData(): pl=19 "media-u4e34f9bu_b128000_195542515.aac"
16:08:30.278 > [0;32m[  6526][I][Audio.cpp:2518] readPlayListData(): pl=20 "#EXTINF:3.008,"m
16:08:30.285 > [  6533][I][Audio.cpp:2518] readPlayListData(): pl=21 "#MY-USER-CHUNK-DATA-1:ON-TEXT-DATA="20240412140821""
16:08:30.295 > 32m[  6544][I][Audio.cpp:2518] readPlayListData(): pl=22 "media-u4e34f9bu_b128000_195542516.aac"
16:08:30.304 > 0;32m[  6560][I][Audio.cpp:2789] parsePlaylist_M3U8(): now playing https://streamcdnc2-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/WjpMtPyNjHwj/#MY-USER-CHUNK-DATA-1:ON-TEXT-DATA="20240412140806"
16:08:32.820 > [0;31m[  9080][E][Audio.cpp:3679] parseHttpResponseHeader(): timeout```

schreibfaul1, 2024-04-12:
The correct URL appears to be http://streamcdnc2-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/chunklist_b128000.m3u8. All others contain video data of different resolutions.   The playlist returned is a bit strange. I have made another change so that the aac files are played. Unfortunately, each aac file contains ~2kB of metadata and the response time from server is very long. This is too much for the ESP32, unfortunately. The stream will not be smooth.

FatherMarco1971, 2024-04-15:
THanks, now it parse m3u8 correctly but, as you say it is not smooth due the metadata. Thanks.

closed 2024-04-15 by FatherMarco1971