schreibfaul1/ESP32-audioI2S #726: HLS

lidense, 2024-04-17:
Hi, is HLS working in this library?
I tried with a bbc stream: 
audio.connecttohost("http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_scotland_fm/bbc_radio_scotland_fm.isml/bbc_radio_scotland_fm-audio%3d48000.norewind.m3u8");
but it does not seem to work. 
Thanks for your help

schreibfaul1, 2024-04-17:
That looks pretty good, what does the log say?

lidense, 2024-04-18:
I have actually succeeded in tuining into that bbc link by downloading a commit that apparently was modified for that particular purpose. It's great, I can no listen to both mp3  and HLS streams with my M5 Core2. This library is great!!!
Is there a documentation that helps in understanding the code it uses in particular for HLS? Thanks a lot!

closed 2024-06-02 by github-actions[bot]