schreibfaul1/ESP32-audioI2S #728: Powersaving potential?

PMKrol, 2024-04-19:
Hi!

I'm building portable audio player. I am wondering is it possible to reduce somehow power consumption.

I was wondering about lowering cpu frequency, but even with 160MHz sound gets choppy. 
I had an idea to do something like this:

```
audio.loop();
setCpuFrequencyMhz(80);
vTaskDelay(1);
setCpuFrequencyMhz(240);
```

but it also does not work - sound gets choppy. I know i2s does some stuff _behind_, but I thought CPU frequency won't affect this. I was wrong, but I do not understand why.
I've checked what happens, and with lower CPU APB works on 80MHz:

```
CPU = 240 MHz, XTAL = 40 MHz, APB = 80000000 Hz
CPU = 160 MHz, XTAL = 40 MHz, APB = 80000000 Hz
CPU = 80 MHz, XTAL = 40 MHz, APB = 80000000 Hz
```

So the problem must be somewhere else. According to https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2s.html (I2S clock section) clock can be sourced from APLL:

> Audio PLL clock, which is more precise than I2S_CLK_SRC_PLL_160M in high sample rate applications. Its frequency is configurable according to the sample rate. However, if APLL has been occupied by EMAC or other channels, the APLL frequency cannot be changed, and the driver will try to work under this APLL frequency. If this frequency cannot meet the requirements of I2S, the clock configuration will fail.

I do not know how ESP32-audioI2s is configured now, but maybe it is the way to reduce some power usage here?

Also, I've found this topic: https://www.esp32.com/viewtopic.php?t=14185, but it is way over my undestanding possibilities ;)

Best regards.

schreibfaul1, 2024-04-19:
I'm not the I2S expert. As I see it, there are only two sources for the clock.
 [image] 
That is the crystal and the PLL.

I2S_CLK_SRC_APLL is in the documentation, I have no idea how to select it. [image]

Best regards.

schreibfaul1, 2024-04-22:
I don't think so. I2S_CLK_SRC_APLL no longer exists in IDF V5.1. (or I can't find it?)

PMKrol, 2024-04-29:
@schreibfaul1 , shameless bump!

SanZamoyski, 2024-05-07:
I do not know where actually I've found stuff that helped, but now I almost can use this in Audio loop:

```
       setCpuFrequencyMhz(80);   
        vTaskDelay(1 / portTICK_RATE_MS);

        if(!audio_pause){
            setCpuFrequencyMhz(240);
        }`
```

it hangs after short while but no choppy sound!

This is what I modified in Audio.cpp:197

```
#else
    m_i2s_config.sample_rate          = 44100; //16000;                                     //PMK
    m_i2s_config.bits_per_sample      = I2S_BITS_PER_SAMPLE_16BIT;
    m_i2s_config.channel_format       = I2S_CHANNEL_FMT_RIGHT_LEFT;
    m_i2s_config.intr_alloc_flags     = ESP_INTR_FLAG_LEVEL1; // interrupt priority
    m_i2s_config.dma_buf_count        = 16;
    m_i2s_config.dma_buf_len          = 512;
    m_i2s_config.use_apll             = 1; //0 // must be disabled in V2.0.1-RC1            //PMK
    m_i2s_config.tx_desc_auto_clear   = true;   // new in V1.0.1
    m_i2s_config.fixed_mclk           = false; //true;                                      //PMK
    m_i2s_config.mclk_multiple        = I2S_MCLK_MULTIPLE_128;
    m_i2s_config.bits_per_chan        = I2S_BITS_PER_CHAN_DEFAULT; // Use bits per sample    //PMK

    if (internalDAC)  {
        #ifdef CONFIG_IDF_TARGET_ESP32  // ESP32S3 has no DAC
        printf("internal DAC");
        m_i2s_config.mode             = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN );
        m_i2s_config.communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_MSB); // vers >= 2.0.5
        i2s_driver_install((i2s_port_t)m_i2s_num, &m_i2s_config, 0, NULL);
        i2s_set_dac_mode((i2s_dac_mode_t)m_f_channelEnabled);
        if(m_f_channelEnabled != I2S_DAC_CHANNEL_BOTH_EN) {
            m_f_forceMono = true;
        }
        #endif
    }
    else {
        // powerdown APLL/PLLA
        SET_PERI_REG_MASK(RTC_CNTL_ANA_CONF_REG, RTC_CNTL_PLLA_FORCE_PD_M);
        CLEAR_PERI_REG_MASK(RTC_CNTL_ANA_CONF_REG, RTC_CNTL_PLLA_FORCE_PU_M);

        m_i2s_config.mode             = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX);
        m_i2s_config.communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_I2S); // Arduino vers. > 2.0.0
        i2s_driver_install((i2s_port_t)m_i2s_num, &m_i2s_config, 0, NULL);
        m_f_forceMono = false;

        // powerup APLL/PLLA
        SET_PERI_REG_MASK(RTC_CNTL_ANA_CONF_REG, RTC_CNTL_PLLA_FORCE_PU_M);
        CLEAR_PERI_REG_MASK(RTC_CNTL_ANA_CONF_REG, RTC_CNTL_PLLA_FORCE_PD_M);
    }
    i2s_zero_dma_buffer((i2s_port_t) m_i2s_num);

```

SanZamoyski, 2024-05-08:
After alot of messing I got this (only important things to show idea:

As setup:
```
    audio.setBufsize(-1, UINT16_MAX * 8);  //didn't help on choppy sound
    loadAudioFile(dirName, fileName);

    int lowCpu = 0;
    unsigned long cpuLowTime = 0;
    unsigned long cpuHighTime = 0;
    unsigned long cpuTimeStart = millis();

    int bufFill = 0;
    int bufSize = /*audio.inBufferSize();*/ UINT16_MAX * 8;
    int queueCmd = 0;
```

"loop" part:
```
//if(audioloaded){
            audio.loop();
            bufFill = audio.inBufferFilled()*100/bufSize;
        //}
        //loopTimes[0][loopRow++] = millis() - loopStart;

        if(audioLowBuf > bufFill){
            audioLowBuf = bufFill;
        } 
        
        //Serial.println((String)bufFill + "%, " + (String)lowCpu);

        if(bufFill < 95 && lowCpu == 1){
            //Serial.println(">");
            cpuLowTime += millis() - cpuTimeStart;
            cpuTimeStart = millis();
            setCpuFrequencyMhz(240);
            lowCpu = 0;
        }

        if(bufFill > 95 && lowCpu == 0){
            //Serial.println("<");
            cpuHighTime += millis() - cpuTimeStart;
            cpuTimeStart = millis();
            setCpuFrequencyMhz(160);
            lowCpu = 1;
        }
```

In debug message I get:
```
17:43:42.770 > [DBG] [BLE] lastNot.: 15 s. ago. Clients�r
17:43:42.775 > �ٹ: 1    [Audio] Buf min.: 0%.   CPU Time Low: 7459, High: 2952, Low ratio: 71%[CPU] freq: 160
17:43:57.839 > [DBG] [BLE] lastNot.: 30 s. ago. Clients: 0. Adv.: 1     [Audio] Buf min.: 94%.  CPU Time Low: 11827, High: 3185, Low ratio: 78%[CPU] freq: 160
```

Output is corrupted due to often change of cpu freq.

But the conception seems to be corrent, is it?
So we get a little powersave. Few mA should be.
But esp32 hangs from time to time, ble does not work.

Is it possible to split PSRAM memory (buffer) filling and DMA stuff?
I'm thinking that everything works at low CPU while buffer is quite filled and only data is put to DMA memory.
Problem is when data from flac is decoded to PSRAM.

Is this correct?

closed 2024-06-23 by github-actions[bot]

PMKrol, 2024-11-07:
OK, got it!

I can now lower the CPU to 160 MHz while playing audio correctly. I'm getting 60-80% low CPU time -> 20-25% powersavings!

> 09:45:07.688 > [DBG] [BLE] lastNot.: 100 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 146390460 (run) [CPU] freq: 160, Low ratio: 70%
> 09:45:37.763 > [DBG] [BLE] lastNot.: 130 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 149863812 (run) [CPU] freq: 160, Low ratio: 74%
> 09:46:07.837 > [DBG] [BLE] lastNot.: 160 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 100% Pos: 151731560 (run) [CPU] freq: 160, Low ratio: 81%
> 09:46:37.902 > [DBG] [BLE] lastNot.: 190 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 155581744 (run) [CPU] freq: 160, Low ratio: 63%
> 09:47:07.953 > [DBG] [BLE] lastNot.: 220 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 159448312 (run) [CPU] freq: 160, Low ratio: 62%
> 09:47:38.025 > [DBG] [BLE] lastNot.: 250 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 163347648 (run) [CPU] freq: 160, Low ratio: 61%
> 09:48:08.086 > [DBG] [BLE] lastNot.: 280 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 167246984 (run) [CPU] freq: 160, Low ratio: 61%
> 09:48:38.189 > [DBG] [BLE] lastNot.: 311 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 171195472 (run) [CPU] freq: 160, Low ratio: 59%
> 09:49:08.257 > [DBG] [BLE] lastNot.: 341 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 174685208 (run) [CPU] freq: 160, Low ratio: 65%
> 09:49:38.314 > [DBG] [BLE] lastNot.: 371 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 176618492 (run) [CPU] freq: 240, Low ratio: 88%
> 09:50:08.391 > [DBG] [BLE] lastNot.: 401 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 100% Pos: 179780548 (run) [CPU] freq: 160, Low ratio: 68%
> 09:50:38.465 > [DBG] [BLE] lastNot.: 431 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 183384972 (run) [CPU] freq: 160, Low ratio: 72%
> 09:51:08.530 > [DBG] [BLE] lastNot.: 461 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 187054932 (run) [CPU] freq: 160, Low ratio: 73%
> 09:51:38.602 > [DBG] [BLE] lastNot.: 491 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 189856540 (run) [CPU] freq: 240, Low ratio: 85%
> 09:52:08.672 > [DBG] [BLE] lastNot.: 521 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 192494336 (run) [CPU] freq: 160, Low ratio: 71%
> 09:52:38.747 > [DBG] [BLE] lastNot.: 551 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 196197064 (run) [CPU] freq: 160, Low ratio: 67%
> 09:53:08.818 > [DBG] [BLE] lastNot.: 581 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 199883408 (run) [CPU] freq: 160, Low ratio: 70%
> 09:53:38.914 > [DBG] [BLE] lastNot.: 611 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 203569752 (run) [CPU] freq: 160, Low ratio: 73%
> 09:54:08.977 > [DBG] [BLE] lastNot.: 641 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 206912032 (run) [CPU] freq: 240, Low ratio: 85%
> 09:54:39.047 > [DBG] [BLE] lastNot.: 671 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 209451524 (run) [CPU] freq: 240, Low ratio: 84%
> 09:55:09.117 > [DBG] [BLE] lastNot.: 702 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 212777420 (run) [CPU] freq: 160, Low ratio: 67%
> 09:55:39.191 > [DBG] [BLE] lastNot.: 732 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 216332692 (run) [CPU] freq: 160, Low ratio: 72%
> 09:56:09.272 > [DBG] [BLE] lastNot.: 762 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 220051804 (run) [CPU] freq: 160, Low ratio: 72%
> 09:56:39.348 > [DBG] [BLE] lastNot.: 792 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 223820068 (run) [CPU] freq: 160, Low ratio: 73%
> 09:57:09.409 > [DBG] [BLE] lastNot.: 822 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 225622280 (run) [CPU] freq: 160, Low ratio: 78%
> 09:57:39.478 > [DBG] [BLE] lastNot.: 852 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 229079248 (run) [CPU] freq: 160, Low ratio: 71%
> 09:58:09.548 > [DBG] [BLE] lastNot.: 882 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 232716440 (run) [CPU] freq: 160, Low ratio: 74%
> 09:58:39.636 > [DBG] [BLE] lastNot.: 912 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 235321440 (run) [CPU] freq: 240, Low ratio: 87%
> 09:59:09.699 > [DBG] [BLE] lastNot.: 942 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 239040552 (run) [CPU] freq: 240, Low ratio: 77%
> 09:59:39.768 > [DBG] [BLE] lastNot.: 972 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 241924108 (run) [CPU] freq: 160, Low ratio: 69%
> 10:00:09.833 > [DBG] [BLE] lastNot.: 1002 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 245741524 (run) [CPU] freq: 160, Low ratio: 62%
> 10:00:39.913 > [DBG] [BLE] lastNot.: 1032 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 249575324 (run) [CPU] freq: 160, Low ratio: 65%
> 10:01:09.983 > [DBG] [BLE] lastNot.: 1062 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 253392712 (run) [CPU] freq: 160, Low ratio: 67%
> 10:01:40.052 > [DBG] [BLE] lastNot.: 1092 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 255325996 (run) [CPU] freq: 160, Low ratio: 56%
> 10:02:10.133 > [DBG] [BLE] lastNot.: 1123 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 258717456 (run) [CPU] freq: 160, Low ratio: 70%
> 10:02:40.217 > [DBG] [BLE] lastNot.: 1153 s. ago. Clients: 0. Adv.: 1 [Audio] Buf min.: 14% Pos: 260503284 (run) [CPU] freq: 240, Low ratio: 88%
[... cut ...]
        CLEAR_PERI_REG_MASK(RTC_CNTL_ANA_CONF_REG, RTC_CNTL_PLLA_FORCE_PD_M);               //PMK
    }
    i2s_zero_dma_buffer((i2s_port_t) m_i2s_num);
```
This is to configure APLL.

in Audio.h I made playAudioData();  public.

@schreibfaul1, this is made locally ofcorse and on some older version of library. Library folder date is 6.05.2024.
Can I ask You to test this solution or, at least, add those modification to official library version? I would like to use current verisions of audioI2S without modyfing it each time...
Maybe something like #define AUDIO_APLL?

Best regards!

schreibfaul1, 2024-11-07:
Hello,
this is a library for universal use. It will never change the CPU frequency.
The fill level of the buffer can be viewed externally, perhaps you can use it to realise your project.

PMKrol, 2024-11-07:
It was not my intention!
My question was only to add a possibility to configure this library for APLL by some swich that configures i2s this way:

```
m_i2s_config.sample_rate          = 44100; //16000;                                     //PMK
    m_i2s_config.bits_per_sample      = I2S_BITS_PER_SAMPLE_16BIT;
    m_i2s_config.channel_format       = I2S_CHANNEL_FMT_RIGHT_LEFT;
    m_i2s_config.intr_alloc_flags     = ESP_INTR_FLAG_LEVEL1; // interrupt priority
    m_i2s_config.dma_buf_count        = 8;      //8
    m_i2s_config.dma_buf_len          = 512;   //512
    m_i2s_config.use_apll             = 1; //0 // must be disabled in V2.0.1-RC1            //PMK
    m_i2s_config.tx_desc_auto_clear   = true;   // new in V1.0.1
    m_i2s_config.fixed_mclk           = false; //true;                                      //PMK
    m_i2s_config.mclk_multiple        = I2S_MCLK_MULTIPLE_128;
    m_i2s_config.bits_per_chan        = I2S_BITS_PER_CHAN_DEFAULT; // Use bits per sample    //PMK
```

and later parts just like `if (internalDAC)  { ...` swich.
And making playAudioData(); public. 

I'll check if modification in processLocalFile()  is really needed.

That way it'll be possible to use APLL intentionally and it wont influence _normal_ usge of library.

If not I'll stick with local library, but I was hoping it can be my (another) contribution in this library evolution ;)