schreibfaul1/ESP32-audioI2S #731: radio using HLS not working

lidense, 2024-04-21:
I would like to listen to this radio:
https://radioitaliasmi.akamaized.net/hls/live/2093120/RISMI/master.m3u8
Works with vlc but not with this library, any ideas?

lidense, 2024-04-22:
Do you think there might be a workaround? It works well with vlc and mpd...

schreibfaul1, 2024-04-22:
When I call up the URL https://radioitaliasmi.akamaized.net/hls/live/2093120/RISMI/master.m3u8, the m3u8 file is transferred and the connection is terminated. The connection remains open for the URLs to the audio sequences, as expected. Unfortunately, the ESP32 takes too long to establish an https connection (it will be fast enough on a PC). If anyone knows a solution, please post here.

lidense, 2024-04-23:
I gathered that the /.../ is a relative link so I put this as the address for audio.connecttohost:
"https://radioitaliasmi.akamaized.net/hls/live/2093120-b/RISMI/stream01/streamPlaylist.m3u8"
And now IT WORKS perfectly!!

schreibfaul1, 2024-04-23:
great! the connection is not terminated after receiving the m3u8 playlist.
The reason for this is that it can take a few seconds to establish a TLS connection. The WiFiClientSecure blocks during this time.
The WiFiClient can be replaced by the HTTPClient without blocking. But there is nothing for WiFiClientSecure and I am not familiar with SSL.

closed 2024-06-07 by github-actions[bot]