schreibfaul1/ESP32-audioI2S #732: Problem with External Audio DAC using the board ESP32-2432S028

didboy, 2024-04-22:
Hi,
I am using this library to get audio through an external DAC type PCM1502a installed on an ESP32-2432S028.

I am trying to configure the pin out of the library for the ESP32-2432S028. I am taking data from this website of macsbug: 
 https://macsbug.wordpress.com/2022/08/20/web-radio-esp32-2432s028-i2s/

Apparently the pin out for this board is: 

#define LRC  27 // are using Set to I2S pin
#define BCLK 22 // are using Set to I2S pin  
#define DIN  17 // are using Set to I2S pin

And in the code I also use the following line to initialize the DAC: 
audio.setPinout(BCLK, LRC, DIN);         // Setup I2S Pin

But I no longer get any audio through the external DAC. 
Am I something wrong? 

PS: ( I also used this other pinout but without success)
#define BCLK 4 // are using Set to I2S pin
#define LRC 16 // are using Set to I2S pin
#define DIN 17 // are using Set to I2S pin
 
Thanks in advance

schreibfaul1, 2024-04-22:
I don't see any reason why this shouldn't work. On some boards the solder bridges are missing:
[image]
or you haven't connected SCK to GND?
[image]

didboy, 2024-04-22:
Hi Screiblfaul1,
:-D 
Many thanks for your quick reply.

Yes. I have jumped the bridge between SCK to GND. 

I have used this same DAC with your library on a standard ESP32 board and worked perfectly.  The issue is on the board ESP32-2432s028.  

I try everything again.  

Thanks again! 
Regards,
Dídac

kenkit, 2024-05-04:
Try using pin 25,26 and 27 also try chaing board to esp32 dev module

didboy, 2024-05-04:
Hi Kensit,
Thanks for your comments on this. 

I tried with this pinout without success. Besides, pin 25 is also required to configure the touch configuration and then the touch function was disabled. 

As soon as I find a solution I let you know. 

Thanks!!

closed 2024-06-18 by github-actions[bot]