schreibfaul1/ESP32-audioI2S #737: Dropouts and choppy sound

Arielhh, 2024-04-29:
Hi, any ide why i am experiencing dropouts in this specific station?
I am using Arduino IDE.

*WM: 
*WM: AutoConnect Try No.:
*WM: 0
*WM: Connecting as wifi client...
*WM: Try to connect with saved credentials
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.177
Current mono mode: Mono
info        Connect to new host: "http://87.120.6.86:8000/hq"
info        PSRAM found, inputBufferSize: 638965 bytes
info        buffers freed, free Heap: 117868 bytes
info        Connection has been established in 137 ms, free Heap: 116944 bytes
info        MP3Decoder has been initialized, free Heap: 118988 bytes , free stack 5856 DWORDs
info        stream ready
info        syncword found at pos 2
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: N/A
info        StreamTitle=''
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible

FatherMarco1971, 2024-04-30:
To work library need buffer is filled fast.
Whatc Better_WiFi_throughput example, you need platformio.

Arielhh, 2024-05-09:
> To work library need buffer is filled fast. Look at Better_WiFi_throughput example, you need platformio.

I don't use PlatformIO. Is there a way to do it in Arduino IDE? Why the buffer size cannot be increased in the library?

FatherMarco1971, 2024-05-10:
To be honest i can't help you, i've reported what suggested by Wolle regarding this issue.
I use platformio but i don't know how to include the Better_WiFi_throughput suggestion in my project...
I've can't listen some web radio because  of the exactly same problem you have reported.

schreibfaul1, 2024-05-10:
Under certain conditions, WiFi is the bottleneck, especially with high bit rates and large data packets. To improve this, you can change the WiFi RX buffer settings in menuconfig. This requires Arduino to run as a component, the precompiled Adruino versions do not allow this.

FatherMarco1971, 2024-05-10:
To me the concept is absolutely clear, i'vent enough knowledge, in platformio and probably in programming, to make it pratically :-)

Arielhh, 2024-05-10:
Thanks Wolle and FatherMarco1971 its all clear now. I guess i will have to use PlatformIO.

closed 2024-05-23 by Arielhh