schreibfaul1/ESP32-audioI2S #741: speech / host does not work together

aber68, 2024-05-08:
hello, why does this lines not work on arduino ESP32-S3? thx 4 help br andreas

... code
**audio.connecttospeech("Radio abc"), "de");   // will never heard
audio.connecttohost(".... radio stream url");**
... code

connecttospeech works only if called without following connecttohost. i want to realize an annoncement of an radio station before playing the stream

schreibfaul1, 2024-05-09:
1) audio.connecttospeech("Radio abc"), "de"); 
2) waiting for audio_eof_speech()
3) audio.connecttohost(".... radio stream url");

aber68, 2024-05-09:
sorry for bother you, but how to code this line #2?

aber68, 2024-05-09:
this will probably work, but i need to choose many radio stations. is there an other possibility to wait on eof_speech? i tried with a flag within audio_eof_speech(const char *info) but this doesn't work

ivuslat, 2024-06-02:
audio.connecttospeech("Radio abc"), "de"); // will never heard

    while (audio.isRunning()) {
    audio.loop();
  }

audio.connecttohost(".... radio stream url");

aber68, 2024-06-02:
thanks, works as well!

closed 2024-07-17 by github-actions[bot]