schreibfaul1/ESP32-audioI2S #744: trouble with some HLS stream

FatherMarco1971, 2024-05-14:
This stream:

http://streamcdnf25-4c4b867c89244861ac216426883d1ad0.msvdn.net/webradio/deejay80/live.m3u8

There a little glitch every time a new TS is loaded (or at least most of the time...)

What is the problem in your opinion?
I don't think is related to wifi optimization.

schreibfaul1, 2024-05-15:
The stream runs well for me with ~60...100KB in the buffer.
The problem is the time it takes to fetch and unpack the new playlist and new data. In the meantime, only the DMA is used. After about 0.3s the DMA is empty and dropouts occur. The cause will be the WiFiclient, we need an asynchronous solution, perhaps the HTTP client. But then https would not work. I don't have any knowledge of encryption.

FatherMarco1971, 2024-05-16:
It is not possible use larger buffer, preload some data and play with a little delay?
I mean you play with 1 second of delay usinga data preloaded...is not clear to me if the library do something like this or not.
And if not...why? This could help with slow stream and in this case too....

schreibfaul1, 2024-05-17:
You cannot make the DMA buffer as large as you like. The current setting is 16x 512 bytes. This is half of the maximum possible size. That is 8KB. For stereo with 2x16bit you need 4 bytes per sample. The usual sample rate is 44.1KHz. How far should the DMA memory reach?

schreibfaul1, 2024-05-17:
This is set here, [image]

if you could put the DMA in PSRAM, that would be cool.

closed 2024-07-01 by github-actions[bot]