schreibfaul1/ESP32-audioI2S #748: Positioning in MP3 player (setAudioPlayPosition & resume pos) not working properly?

Gloeidraad, 2024-05-17:
Hi,

This is the first time ever, I post something on a Github issues list, so I hope I am not conflicting the etiquette. Currently I am including your ESP32-audioI2S library into a project. Most things work perfectly, but I have some issues with the SD card player. In my application, it is important to resume a file at the right location after powering down. Also, I regularly want do move to a different time into a track. I noticed that this seems not to work properly. I few seconds off is no problem, but the current playing time, and the total playing time appear a bit "jumpy". There were some previous remarks about resuming, but I couldn't find a matching one.

So, I created track with audio time stamps, by converting the YouTube track https://www.youtube.com/watch?v=XY8bdG-U26Q to MP3.

This track lasts 5 minutes, (but the poster has also tracks of 10 minutes and 25 minutes).

If I start the track, it will play normally for 5 minutes with a playing time of 5 minutes reported from getAudioFileDuration(). However, if I:

1)
Pause the track at one minute, and then resume, the getAudioFileDuration() reports a (much) longer time, which over time will slowly trickle back to 5:00. This is not a big problem, but a bit confusing.

2)
The same issue appears when I reposition inside the track. I play to about one minute. Then I pause the player and call setAudioPlayPosition(180) to skip to three minutes, the end time also changes. Also not a big problem.

3)
This one is a bigger problem. When I want to turn off the player, I store the current location:

NV.DiskTrackResumeTime = _audio->getAudioCurrentTime();
NV.DiskTrackResumePos  = _audio->getFilePos();
NV.DiskTrackTotalTime  = _audio->getAudioFileDuration();

(NV being some non-volatile memory).

If I turn on the player, these values are used to resume the track at the same location. The two time settings are used on the display for a nice look. The DiskTrackResumePos is used to resume playing at the desired location:

_audio->connecttoFS(SD, track.c_str(), NV.DiskTrackResumePos); 

However, this does not work. The resume position is about 20-30 seconds ahead of the expected location. In addition the total playing time also changes to a much longer time. I did some investigation. I stopped the track at 71 seconds. The reported track length was at that moment 308 seconds. This means that the track was about 23% in play. However, the file position returned 2331836, with a start position of 138 and a file size of 7213067. With some calculations, I determined that the audio track should then be about 32% progressed. In playing time, this is 99 seconds, which is indeed close to the time appeared on the display. So, it looks like the file was progressed a lot further then the actual playing time. Perhaps due to buffering (I use a WROVER with PSRAM enabled).

As a workaround I now calculate the resume position from the playing times:

NV.DiskTrackResumePos  = _audio->getAudioDataStartPos() + (_audio->getFileSize() - _audio->getAudioDataStartPos()) * NV.DiskTrackResumeTime / NV.DiskTrackTotalTime;

This seems to work reasonably, but I am not sure, if I used all the correct parameters. Or perhaps I misunderstood the meaning of getFilePos()?

Regards,
Otto

schreibfaul1, 2024-05-17:
Hello Otto,
it is not easy to determine the times. With CBR it is still ok. VBR is estimated. It would be better to take samples from the file at several points, find the beginning of the mp3 frames and then determine the bit rate and calculate the BR from that.  But this would take too long with the ESP32.
There are two files for measuring:  [VBR](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/additional_info/Testfiles/seconds-CBR.mp3) and [CBR](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/additional_info/Testfiles/seconds-CBR.mp3)

The size of the audio block can be determined with **getFileSize()** and **getAudioDataStartPos()**. This is ok in your equation, metadata and embedded images are calculated out.
**getFilePos()** returns the pointer to the audio file. This is not the position that is currently being played, but the position from which the next data will be fetched. To calculate the actual play position, the fill level of the buffer must be taken into account. This is then **getFilePos() - inBufferFilled()**

Gloeidraad, 2024-05-17:
Thanks Wolle,
I suspected it had something to do with the buffer, but I missed the usage of  inBufferFilled. In the case I mentioned above the buffer was filled with 638338 bytes. If I recalculate the size played by subtracting this number for the file position, I come very close to the number of my previous calculation. Thanks for your feedback...

Gloeidraad, 2024-05-17:
[image]

By the way, there was one song which wouldn't play. It didn't start up, and locked up the ESP32 completely. I needed to power cycle the unit.  I investigated the file with Mp3tag (see image above), and noticed that the embedded cover image was very large (3MB). I removed the cover image, and after that the track played normally. Could it be that the cover image may not be loo large?

schreibfaul1, 2024-05-18:
I have been able to improve the speed a little. The APIC image size can now be a maximum of about 6.5MB without triggering a timeout.

Gloeidraad, 2024-05-18:
After I use setAudioPlayPosition, connecttoFS (perhaps also in setTimeOffset), the current play time jumps around a few seconds, before stabilizing. In setAudioPlayPosition(uint16_t sec) I added the line m_audioCurrentTime = sec;, which stops the jumping around. Perhaps it is also possible in connecttoFS with resume position, to make a (crude) estimation of the matching time.

Gloeidraad, 2024-05-19:
Hi, I did some additional testing on the reported times. It is OK for my application, hence I deleted my previous post. The track which wouldn't play, does still not play. I guess you did not addressed that issue. Thanks anyway for your kind feedback.

Gloeidraad, 2024-05-26:
Hi Wolle,

I have to come  back at the issue of not starting MP3 tracks. On an SD card, I have a test set of 45 MP3 tracks. Normally, I use only a few of the first tracks to test if the player is running properly. After I discovered the non working track, you changed the line:

`availableBytes = 256 * 1024; // set some large value`

Then I discovered, that three more tracks did occasionally not start playing. So I created a very basic player to rule out interference with my application. Some of the tracks do start when I go back to

`availableBytes = 16 * 1024; // set some large value`

And one only started when going to:

`availableBytes = 8 * 1024; // set some large value`

Then I decided to turn off PSRAM. Then the tracks will play with 256 * 1024 size. 

I use "ESP32 Wrover Kit (all versions)" with this sketch:

```
// will play many audio file formats, mp3,aac,flac etc.
// See github page : https://github.com/schreibfaul1/ESP32-audioI2S
 
#include "Arduino.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"
 
// Digital I/O used
#define SD_CS          5
#define SPI_MOSI      23    // SD Card
#define SPI_MISO      19
#define SPI_SCK       18
 
#define I2S_DOUT      12
#define I2S_BCLK      27    // I2S
#define I2S_LRC       33
 
Audio audio;
 
void setup() {
    Serial.begin(115200);
    pinMode(SD_CS, OUTPUT);      
    digitalWrite(SD_CS, HIGH);
    delay(500);
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    if(!SD.begin(SD_CS))
    {
      Serial.println("Error talking to SD card!");
      while(true);  // end program
    }
    else
      Serial.println("Start Playing: ");
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setI2SCommFMT_LSB(true);  // For PT8211
    audio.setVolume(15); // 0...21
    audio.connecttoFS(SD,"track name.mp3"); // Change to a valid name
}
 
void loop()
{
    audio.loop();    
}

```

schreibfaul1, 2024-05-27:
Hi Gloeidraad, 
that's exactly what I changed so that embedded images are skipped faster. I'm using the same sketch with SD_MMC though. I don't know what could be the cause, I can't reproduce the error. Do you see anything unusual in the log?

Gloeidraad, 2024-05-27:
> Hi Gloeidraad, that's exactly what I changed so that embedded images are skipped faster. I'm using the same sketch with SD_MMC though. I don't know what could be the cause, I can't reproduce the error. Do you see anything unusual in the log?

I am not so experienced into Arduino. Do you mean the compile log?
I do not use the MMC. I thought about it, but it seems that you cannot change the I/O pins for that. To test, I have to patch a bit in my design. I will try, but it may take a while.

Gloeidraad, 2024-05-31:
I figured out, I had to set the "Core Debug Level" to "Verbose" for more information. On the the non-playing tracks, I got:

`[  3674][E][Audio.cpp:3097] processLocalFile(): audioHeader reading timeout`

So, in processLocalFile() I played a bit around with:

`const uint32_t  timeout = 2500;  // ms`

For the file which never played, I had to increase the timeout to a staggering 8600 ms to to get rid of the timeout. I found out, that the default SPI speed speed is only 4 MHz. So, I changed the speed to 25 MHz by changing:

`if(!SD.begin(SD_CS)) {...}`

to

`if(!SD.begin(SD_CS, SPI, 25000000)) {...}`

Now the track plays with a timeout of "only" 3200 ms. However, I do not know if all micro SD cards are capable of handling a SPI speed of 25 MHz. Perhaps a higher frequency is even possible and allowed. 

Then, I rewired the interface to MMC. This interface appears to work faster, but the problem track still timed out. However, I only needed to increase the timeout to 2700 to let it play. Perhaps you can make this (or maybe also some other timeouts) adjustable, like:

`bool setHeaderTimeout(int ms);`

_By the way, I do not know why one track originally played with:_

`availableBytes = 8 * 1024;`

_but timed out with_

`availableBytes = 16 * 1024;  (or 256 * 1024)`

_Perhaps some kind of memory allocation optimization?_

schreibfaul1, 2024-05-31:
Thank you for your contribution. It's really just a matter of finding the beginning of the audio block and skipping all irrelevant data. This also depends on how much data can be read from the SD card at once.
You can operate today's standard SD cards at 80MHz.
'SD_MMC.begin("/sdcard", true, false, sdmmc_frequencyKHz)'
Irrelevant data should not even be read, but should be skipped immediately. Unfortunately, this does not always work for all file formats.

closed 2024-07-16 by github-actions[bot]