schreibfaul1/ESP32-audioI2S #752: MP3 Decoder Error

AntoineLep, 2024-05-23:
Hello,

I am trying to read an MP3 file stored inside an SD card in an ESP32.
The file is played correctly for around 1 minute and then I have the following errors popping off my serial monitor:

```info        Audio-Length: 31681306
info        stream ready
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: 320000
info        MP3 decode error -6 : INVALID_FRAMEHEADER
info        syncword found at pos 137
info        syncword found at pos 0
info        MP3 decode error -6 : INVALID_FRAMEHEADER
info        syncword found at pos 0
info        MP3 decode error -6 : INVALID_FRAMEHEADER
info        syncword found at pos 0
info        MP3 decode error -6 : INVALID_FRAMEHEADER
info        syncword found at pos 0
info        MP3 decode error -6 : INVALID_FRAMEHEADER
info        syncword found at pos 0
info        MP3 decode error -6 : INVALID_FRAMEHEADER
info        syncword found at pos 0
info        MP3 decode error -6 : INVALID_FRAMEHEADER```

Any idea on how I can handle this ?
There is no problem when reading it from a computer.

From what I see it looks like something is not correctly decoded in helix-mp3 decoder and the Audio is not fault tolerant like in a computer player.

Thanks

schreibfaul1, 2024-05-23:
Is this also the case with other mp3 files? That is difficult to judge. The mp3 header follows the sync word and tells the decoder how to decode the data. There seems to be something out of specification here.
Do you have the possibility to insert the mp3 file or a link here?

AntoineLep, 2024-05-23:
Yes, you can download it here (valid for one hour): https://shorturl.at/6voLM
It cut at 77 sec
Some other files are working correctly but not this one

schreibfaul1, 2024-05-23:
The problem is probably somewhere else. I have downloaded the file and it runs through the 13:12 min without any problems.

AntoineLep, 2024-05-23:
Did you test it using an SD card (FAT32) on an ESP32 (Wrover Module) as well ?

I can share you my code if it can help reproduce:

```c++
#include "Arduino.h"
#include "SD.h"
#include "FS.h"
#include "SPI.h"
#include "Audio.h"

#define PW GPIO_NUM_21
#define GAIN GPIO_NUM_23
#define SDD GPIO_NUM_34  // SD card module

#define SD_CS 13
#define SPI_MOSI 15
#define SPI_MISO 2
#define SPI_SCK 14
#define I2S_DOUT 26
#define I2S_BCLK 5
#define I2S_LRC 25
#define I2S_DIN 35
#define SDA 18
#define SCL 23

#define DEFAULT_VOLUME 5

// Method declarations
void audioTask(void* parameters);
void controlInputTask(void* parameters);
void togglePlayPause();
void handleForward();
void handleBackward();
void updateAudioInfo();

// Create instances for MP3 playback
Audio audio;

// Player data
int currentTime = 0;
bool isPlaying = false;
int volume = DEFAULT_VOLUME;

void setup() {
  Serial.begin(115200);

  // Configure SPI
  Serial.println("Configuring SPI");
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);

  // Initialize SD card
  Serial.println("Mounting card");
  if (!SD.begin(SD_CS)) {
    Serial.println("SD Card Mount Failed");
    return;
  }
  Serial.println("SD Card Initialized");

  // Amp power enable
  gpio_reset_pin(PW);
  gpio_set_direction(PW, GPIO_MODE_OUTPUT);
  gpio_set_level(PW, 1);

  gpio_reset_pin(GAIN);
  gpio_set_direction(GAIN, GPIO_MODE_OUTPUT);
  gpio_set_pull_mode(GAIN, GPIO_PULLDOWN_ONLY);

  // Initialize I2S
  Serial.println("Initializing I2S");
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  Serial.println("I2S installed and configured");

  audio.setVolume(volume);  // 0..21

  // Create tasks
  xTaskCreatePinnedToCore(
    audioTask,     // Task function
    "Audio Task",  // Task name
    8192,          // Stack size (in words)
    NULL,          // Task input parameter
    1,             // Task priority
    NULL,          // Task handle
    0              // Core to run the task on (0 or 1)
  );

  xTaskCreatePinnedToCore(
    controlInputTask,      // Task function
    "Control Input Task",  // Task name
    2048,                  // Stack size (in words)
    NULL,                  // Task input parameter
    1,                     // Task priority
    NULL,                  // Task handle
    1                      // Core to run the task on (0 or 1)
  );
}

void audioTask(void* parameters) {
  // Open the MP3 file from SD card
  audio.connecttoFS(SD, "/audiobooks/6179d79c-a40c-427d-b04a-59ef7ed33cc6/001-Stupeur_et_tremblements.mp3");

  while (true) {
    updateAudioInfo();

    if (isPlaying) {
      audio.loop();
    }

    vTaskDelay(5 / portTICK_PERIOD_MS);  // Delay to allow other tasks to run
  }
}

void controlInputTask(void* parameters) {
  while (true) {
    // Check for serial input to toggle play/pause
    if (Serial.available() > 0) {
      char command = Serial.read();
      switch (command) {
        case 'p':
          togglePlayPause();
          break;
        case 'f':
          handleForward();
          break;
        case 'b':
          handleBackward();
          break;
      }
    }
    vTaskDelay(100 / portTICK_PERIOD_MS);  // Delay to allow other tasks to run
  }
}

void togglePlayPause() {
  audio.pauseResume();
}

void handleForward() {
  Serial.println("handleForward");
}

void handleBackward() {
  Serial.println("handleBackward");
}

void updateAudioInfo() {
  if (audio.isRunning()) {
    if (!isPlaying) {
      isPlaying = true;
      Serial.println("Now playing");
    }

    if (audio.getAudioCurrentTime() != currentTime) {
      currentTime = audio.getAudioCurrentTime();
      Serial.print("Audio time: ");
      Serial.println(currentTime);
    }
  } else {
    if (isPlaying) {
      isPlaying = false;
      Serial.println("Now Stopped");
    }
  }
}

void loop() {
  // The loop function is intentionally left empty as tasks are running on both cores
}

void audio_info(const char* info) {
  Serial.print("INFO:        ");
  Serial.println(info);
}

void audio_id3data(const char* info) {
  Serial.print("ID3DATA:     ");
  Serial.println(info);
}

void audio_eof_mp3(const char* info) {
  Serial.print("MP3 EOF:     ");
  Serial.println(info);
}

void audio_bitrate(const char* info) {
  Serial.print("BITRATE:     ");
  Serial.println(info);
}

void audio_icyurl(const char* info) {
  Serial.print("ICYURL:      ");
  Serial.println(info);
}
```

Hope it helps.

schreibfaul1, 2024-05-23:
My SD card is configured as SD_MMC. I don't think SD or SD_MMC makes the difference.
In your example you communicate between the tasks without MessageQueue. That is dangerous.

AntoineLep, 2024-05-24:
You're right, I'll put a Message Queue. Any other recommendations ?

I double checked the files on the SD card and for some reason the problematic one couldn't be read from the SD card in my computer.
Re uploading the exact same fixes it. I don't know what went bad during the first upload…

Thanks for your time !

schreibfaul1, 2024-05-24:
Very good!, here is an example of intertask communication. [audiotask,](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/separate_audiotask/separate_audiotask.ino)
You would need to add some more functions, but it will certainly save some time.
best regards
Wolle

closed 2024-05-27 by AntoineLep