schreibfaul1/ESP32-audioI2S #753: Buffers freed

Rene1695, 2024-05-28:
Hi, 

i have some issue with the library. I am using an ESP32-S3 dev kit 1. I have connected three buttons to the ESP and I simply would like to change the station based on the button press.

Once I start the ESP, audio is playing well (I connect to a station in the setup). If I press some button, station change is working for 1 ... 3 attempts and the audio is collapsing and I can not do anything. In the output I got this 👍 

> info        Connect to new host: "https://radio21.streamabc.net/radio21-bremen-mp3-192-7011720?sABC=664o5s20%230%23nr69r2175702497q02q3s599662q7735%23jro&aw_0_1st.playerid=web&amsparams=playerid:web;skey:1716215584"
info        PSRAM not found, inputBufferSize: 13951 bytes
info        buffers freed, free Heap: 216028 bytes
info        SSL has been established in 368 ms, free Heap: 174416 bytes
bitrate     192000
station     RADIO 21 - Bremen
info        MP3Decoder has been initialized, free Heap: 150740 bytes , free stack 3836 DWORDs
lasthost    https://radio21.streamabc.net/radio21-bremen-mp3-192-7011720?sABC=664o5s20%230%23nr69r2175702497q02q3s599662q7735%23jro&aw_0_1st.playerid=web&amsparams=playerid:web;skey:1716215584
info        stream ready
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: 192000
info        StreamTitle='RADIO 21 - Bremen'
streamtitle RADIO 21 - Bremen
info        StreamTitle='Foreigner - Cold As Ice'
streamtitle Foreigner - Cold As Ice
info        Connect to new host: "https://icecast.radiobremen.de/rb/bremenvier/live/mp3/128/stream.mp3"
info        buffers freed, free Heap: 214548 bytes

It says `buffers freed, free Heap: 214548 bytes`. so I guess there is some kind of issue with some buffer? Can I increase the size of the buffer? What am I doing wrong? 
```
// Additional Librarys
#include <Audio.h>
#include "WiFi.h";
#include <U8g2lib.h>
#include <Wire.h>

// Encoder & Buttons
#define PHASE_A_PIN 12
#define PHASE_B_PIN 13
#define SWITCH_PIN 11
#define BUTTON_1_PIN 7
#define BUTTON_2_PIN 18
#define BUTTON_3_PIN 6
#define MUTE_BUTTON_PIN 46

// External DAC
#define I2S_DOUT 15  // connect to DAC pin DIN
#define I2S_BCLK 16  // connect to DAC pin BCK
#define I2S_LRC  17  // connect to DAC pin LCK

// I2C Bus 1: OLED
#define OLED_ADDR 0x3C
#define OLED_SCK 4
#define OLED_SDA 5

// I2C Bus 2: MAX9744
#define MAX9744_ADDR 0x4B
#define MAX9744_SCL 9
#define MAX9744_SDA 10

// Encoder variables + debouncing
volatile int8_t enc_delta; 
unsigned long lastSwitchTime = 0;
const unsigned long debounceDelay = 200;

// Lautstärke 
volatile int8_t volume = 15;
volatile int8_t last_volume = 0;
bool lastMuteState = true;

// Wifi credentials
const char* ssid = "xxxx";
const char* pwd = "xxxx";
bool wifiConnected = false;

volatile int8_t current_state = 2;

// Radio stations
const char* bremen4 = "https://icecast.radiobremen.de/rb/bremenvier/live/mp3/128/stream.mp3";
const char* radio21 = "https://radio21.streamabc.net/radio21-bremen-mp3-192-7011720?sABC=664o5s20%230%23nr69r2175702497q02q3s599662q7735%23jro&aw_0_1st.playerid=web&amsparams=playerid:web;skey:1716215584";
const char* energy = "http://ffn-stream29.radiohost.de/energybremen_mp3-192?_art=dD0xNzE2MjE1ODUyJmQ9NDcxMzhiMjlkODRlMWUxMTg5ZDk";

// Instance for webradio
Audio audio; 

// Init OLED
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, OLED_SCK, OLED_SDA);

void displayWarning() {

  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_crox1hb_tf); // Font anpassen
  u8g2.setCursor(0, 30); // Position des Textes
  u8g2.print("Wifi nicht verbunden...");
  u8g2.sendBuffer();
}

void isr()
{
  static int8_t last;
  int8_t new_val, diff;

  new_val = 0;
  if (digitalRead(PHASE_A_PIN))
    new_val = 3;
  if (digitalRead(PHASE_B_PIN))
    new_val ^= 1; // convert gray to binary
  diff = last - new_val; // difference last - new
  if (diff & 1)
  {                 // bit 0 = value (1)
    last = new_val;     // store new as next last
    enc_delta += (diff & 2) - 1; // bit 1 = direction (+/-)
  }
}

int8_t encode_read4()
{
  int8_t val;

  noInterrupts();
  val = enc_delta;
  enc_delta &= 3;
  interrupts();
  return val >> 2;
}

void setup()
{
  Serial.begin(115200);
  // Define Pins and Interrupts
  pinMode(PHASE_A_PIN, INPUT_PULLUP);
  pinMode(PHASE_B_PIN, INPUT_PULLUP);
  pinMode(SWITCH_PIN, INPUT_PULLUP);
  pinMode(BUTTON_1_PIN, INPUT_PULLUP);
  pinMode(BUTTON_2_PIN, INPUT_PULLUP);
  pinMode(BUTTON_3_PIN, INPUT_PULLUP);
  pinMode(MUTE_BUTTON_PIN, INPUT_PULLUP);

  attachInterrupt(digitalPinToInterrupt(PHASE_A_PIN), isr, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PHASE_B_PIN), isr, CHANGE);

  // Default Display screen
  u8g2.begin();
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_helvB18_tf);
  u8g2.drawStr(0, 24, "RADIO21");
  u8g2.sendBuffer();

  // Init Wifi
  WiFi.enableSTA(true);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pwd);

  while (WiFi.status() != WL_CONNECTED) 
  {
    displayWarning();
    delay(1000);
  }

  WiFi.setAutoReconnect(true);

    // Setup I2S bus and audio library
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.connecttohost(radio21);
  audio.setVolume(21);

  // Set audio value on MAX9744
  Wire1.setPins(MAX9744_SDA, MAX9744_SCL);
  Wire1.begin();
  Wire1.beginTransmission(MAX9744_ADDR);
  Wire1.write(volume);
  Wire1.endTransmission();
}

void loop(){

  // ENERGY BREMEN
  if (digitalRead(BUTTON_1_PIN) == LOW)
  {
    audio.connecttohost(energy);
    delay(debounceDelay);
  }

  // RADIO 21
  if (digitalRead(BUTTON_2_PIN) == LOW)
  {
    audio.connecttohost(radio21);
    delay(debounceDelay);
  }

  // BREMEN 4
  if (digitalRead(BUTTON_3_PIN) == LOW)
  {
    audio.connecttohost(bremen4);
    delay(debounceDelay);
  }

  audio.loop();

 // wifi_check();
}       

void wifi_check(){

  if(WiFi.status() != WL_CONNECTED) 
  {
    displayWarning();
    audio.setVolume(0);
    delay(1000);
    wifiConnected = false;
  }
  else
  {
    if (!wifiConnected) 
    {
      wifiConnected = true;
      audio.setVolume(21);
    }
  }
}

void audio_info(const char *info) {
  Serial.print("info        ");
  Serial.println(info);
}
void audio_id3data(const char *info) {  //id3 metadata
  Serial.print("id3data     ");
  Serial.println(info);
}
void audio_eof_mp3(const char *info) {  //end of file
  Serial.print("eof_mp3     ");
  Serial.println(info);
}
void audio_showstation(const char *info) {
  Serial.print("station     ");
  Serial.println(info);
}
void audio_showstreamtitle(const char *info) {
  Serial.print("streamtitle ");
  Serial.println(info);
}
void audio_bitrate(const char *info) {
  Serial.print("bitrate     ");
  Serial.println(info);
}
void audio_commercial(const char *info) {  //duration in sec
  Serial.print("commercial  ");
  Serial.println(info);
}
void audio_icyurl(const char *info) {  //homepage
  Serial.print("icyurl      ");
  Serial.println(info);
}
void audio_lasthost(const char *info) {  //stream URL played
  Serial.print("lasthost    ");
  Serial.println(info);
}
void audio_eof_speech(const char *info) {
  Serial.print("eof_speech  ");
  Serial.println(info);
}

```
[image]

schreibfaul1, 2024-05-29:
This is just a guess, but it could lead to multiple calls:

````c++
  // ENERGY BREMEN
  if (digitalRead(BUTTON_1_PIN) == LOW)
  {
    audio.connecttohost(energy);
    delay(debounceDelay);
  }
````
would be better:

````c++
  static bool buttonPressed = false;
  // ENERGY BREMEN
  if (digitalRead(BUTTON_1_PIN) == LOW)  {
    if(!buttonPressed){ 
        audio.connecttohost(energy);
        delay(debounceDelay);
    }
    buttonPressed = true;
  }
else{
    buttonPressed = false;
}
````

Rene1695, 2024-05-29:
Hi @schreibfaul1 thanks for your reply! I did your changes but found that it made no difference (but of course I will let it as you suggested, it's obviously better).

I found out, that if I wait a couple of seconds I got a "bad request" in the serial monitor. After this I can press again on a button and try to connect to a station. 

```
streamtitle RADIO 21 - Bremen
info        StreamTitle='Eagles - Tequila Sunrise'
streamtitle Eagles - Tequila Sunrise
info        StreamTitle='Don Henley - The Boys Of Summer'
streamtitle Don Henley - The Boys Of Summer
info        Connect to new host: "https://icecast.radiobremen.de/rb/bremenvier/live/mp3/128/stream.mp3"
info        buffers freed, free Heap: 214136 bytes
info        Request https://icecast.radiobremen.de/rb/bremenvier/live/mp3/128/stream.mp3 failed!
station     
streamtitle 
icyurl      
ROUTINE3
info        Connect to new host: "http://ffn-stream29.radiohost.de/energybremen_mp3-192?_art=dD0xNzE2MjE1ODUyJmQ9NDcxMzhiMjlkODRlMWUxMTg5ZDk"
info        buffers freed, free Heap: 214508 bytes
info        Request http://ffn-stream29.radiohost.de/energybremen_mp3-192?_art=dD0xNzE2MjE1ODUyJmQ9NDcxMzhiMjlkODRlMWUxMTg5ZDk failed!
```

I played  a little bit around and the error is not really reproducible. Sometimes, after boot up I can change stations within the first seconds freely. After this the above messages starts. Sometimes (if the ESP is on for some time) I can switch between the different stations (and I can do that 15-20 times through all different stations). Sometimes I can not switch to any station. I press different stations and none of them will connect again.

At first I thought it was some issue with the stream link, but as I said, sometimes I can switch between them without any issue. Do you think this might be related to some kind of wifi connection issue? I am not that close to my access point so the only thing I could imaging is that sometimes the connection is good and sometimes it is bad. I am a little bit desperate and I feel like very very stupid :D

Rene1695, 2024-05-29:
I tested the connection with a hotspot from my iPhone right next to the esp but got the same behavior :(

Rene1695, 2024-05-29:
I tried some minimal code, this is my current setup. I simply switching to a new station every 5 seconds without any buttons and stuff. 

``` c
#include <Audio.h>
#include "WiFi.h";

#define I2S_DOUT 15  // connect to DAC pin DIN
#define I2S_BCLK 16  // connect to DAC pin BCK
#define I2S_LRC  17  // connect to DAC pin LCK

const char* bremen4 = "https://icecast.radiobremen.de/rb/bremenvier/live/mp3/128/stream.mp3";
const char* radio21 = "https://radio21.streamabc.net/radio21-bremen-mp3-192-7011720?sABC=664o5s20%230%23nr69r2175702497q02q3s599662q7735%23jro&aw_0_1st.playerid=web&amsparams=playerid:web;skey:1716215584";
const char* energy = "http://ffn-stream29.radiohost.de/energybremen_mp3-192?_art=dD0xNzE2MjE1ODUyJmQ9NDcxMzhiMjlkODRlMWUxMTg5ZDk";

Audio audio; 

unsigned long previousMillis = 0;  // speichert, wann der letzte Wechsel stattfand
const long interval = 5000;        // Intervall von 5 Sekunden (5000 Millisekunden)
int streamIndex = 0;               // Index zum Tracken des aktuellen Streams

const char* ssid = "XXX";
const char* pwd = "XXX";

void setup() {

  Serial.begin(115200);  // Starte die serielle Kommunikation

  WiFi.enableSTA(true);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pwd);

  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(1000);
  }

  WiFi.setAutoReconnect(true);

}

void loop() {
   unsigned long currentMillis = millis();

  // Überprüfe, ob das Intervall abgelaufen ist
  if (currentMillis - previousMillis >= interval) 
  {
    previousMillis = currentMillis;  // Aktualisiere die letzte Wechselzeit

    // Wechsle zum nächsten Stream
    switch (streamIndex) {
      case 0:
        audio.connecttohost(energy);
        streamIndex = 1;
        break;
      case 1:
        audio.connecttohost(radio21);
        streamIndex = 2;
        break;
      case 2:
        audio.connecttohost(bremen4);
        streamIndex = 0;
        break;
    }
  }
  audio.loop();
}

void audio_info(const char *info) {
  Serial.print("info        ");
  Serial.println(info);
[... 117 lines cut ...]
station     RADIO 21 - Bremen
info        MP3Decoder has been initialized, free Heap: 160100 bytes , free stack 3212 DWORDs
lasthost    https://radio21.streamabc.net/radio21-bremen-mp3-192-7011720?sABC=664o5s20%230%23nr69r2175702497q02q3s599662q7735%23jro&aw_0_1st.playerid=web&amsparams=playerid:web;skey:1716215584
info        stream ready
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: 192000
info        StreamTitle='RADIO 21 - Bremen'
streamtitle RADIO 21 - Bremen
info        slow stream, dropouts are possible
info        Connect to new host: "https://icecast.radiobremen.de/rb/bremenvier/live/mp3/128/stream.mp3"
info        buffers freed, free Heap: 223696 bytes
info        SSL has been established in 315 ms, free Heap: 178668 bytes
info        redirect to new host "https://f141.rndfnk.com/ard/rb/bremenvier/live/mp3/128/stream.mp3?cid=01FC1W6QK48EBYZ3RJW932FC1E&sid=2h9d6Qtt9jstLP3Q3QT3kdhoGAE&token=Zh6Sao50U0X0TToh9fpb35z_RRaDpCtNAWy4nqQSMis&tvf=gBWGkjAf1BdmMTQxLnJuZGZuay5jb20"
info        Connect to new host: "https://f141.rndfnk.com/ard/rb/bremenvier/live/mp3/128/stream.mp3?cid=01FC1W6QK48EBYZ3RJW932FC1E&sid=2h9d6Qtt9jstLP3Q3QT3kdhoGAE&token=Zh6Sao50U0X0TToh9fpb35z_RRaDpCtNAWy4nqQSMis&tvf=gBWGkjAf1BdmMTQxLnJuZGZuay5jb20"
info        buffers freed, free Heap: 223072 bytes
info        Request https://f141.rndfnk.com/ard/rb/bremenvier/live/mp3/128/stream.mp3?cid=01FC1W6QK48EBYZ3RJW932FC1E&sid=2h9d6Qtt9jstLP3Q3QT3kdhoGAE&token=Zh6Sao50U0X0TToh9fpb35z_RRaDpCtNAWy4nqQSMis&tvf=gBWGkjAf1BdmMTQxLnJuZGZuay5jb20 failed!
station     
streamtitle 
icyurl      
info        Connect to new host: "http://ffn-stream29.radiohost.de/energybremen_mp3-192?_art=dD0xNzE2MjE1ODUyJmQ9NDcxMzhiMjlkODRlMWUxMTg5ZDk"
info        buffers freed, free Heap: 223612 bytes
```

schreibfaul1, 2024-05-30:
I have just tested this, V3.0.0 in the Arduino IDE. After a short time, the hostname is no longer resolved, regardless of whether ESP32 or ESP32-S3.
````
[158270][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'mp3.ffh.de' with error '-54'
[158532][I][NetworkClient.cpp:253] connect(): select returned due to timeout 250 ms for fd 48
audio_info: Request http://mp3.ffh.de/radioffh/hqlivestream.aac failed!
````
 I can't do much about it, Espressif has some open issues with IP4 and DNS.

Rene1695, 2024-05-30:
Damn :( but thank you for checking on your system!

Rene1695, 2024-05-30:
Do you think if I would try an earlier version (2.x) it might work? Of course this is just temporarily until this is fixed by espressif

schreibfaul1, 2024-05-30:
Yes, it will work with an earlier version. Unfortunately, I can't give a detailed answer to the problem because I'm not very familiar with network issues.

Rene1695, 2024-05-30:
Okey thank you very much :)

closed 2024-05-30 by Rene1695