schreibfaul1/ESP32-audioI2S #755: Arduino 3.0.0 Error 263

rvbc1, 2024-05-29:
Hi,

Today I was testing the new release of stable Arduino 3.0.0. When I uploaded the code, the audio plays (but is not clear), and I also see message spam on the serial port:

```

[140469][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140475][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140498][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140519][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140537][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140544][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140565][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140575][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140590][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140600][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140621][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140636][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140646][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140661][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140672][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140681][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140702][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140725][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140749][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140765][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140775][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140785][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140809][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140830][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140846][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
[140856][E][Audio.cpp:4799] playSample(): ESP32 Errorcode 263
```

I’m using the ESP32 Wrover on a custom board. Here is my PlatformIO configuration:

```
[env]
platform = https://github.com/platformio/platform-espressif32.git#v6.7.0

platform_packages=
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.0/esp32-arduino-libs-3.0.0.zip

board = esp32dev
framework = arduino

lib_deps = 
	https://github.com/schreibfaul1/ESP32-audioI2S.git#3.0.8
	
build_flags = 
	-std=gnu++17
	-DBOARD_HAS_PSRAM 
	-mfix-esp32-psram-cache-issue
	
build_unflags = 
	-std=gnu++11
	-std=gnu99
```

rvbc1, 2024-05-29:
Without these lines:
```
platform_packages=
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.0/esp32-arduino-libs-3.0.0.zip
```
the Arduino version is 2.0.16 (as in PlatformIO 6.7.0), and everything works fine with the same code.

AJR1010, 2024-05-29:
Probably the same problem here. I'm using the Arduino IDE with esp32 platform.
The following minimal code runs fine (switches between two streams, playing each for a few seconds) using the esp32 platform 2.0.17. When I switch to platform 3.0.0 the 2nd stream hangs (see 2nd printout below). I'm using a ESP32-WROVER-E module which has a ESP32-DOWD-V3 dual-core chip and PSRAM.

```
#include <Arduino.h>
#include <WiFi.h>
#include "Audio.h"

#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26

Audio audio;

String ssid =     "___";
String password = "___";

String stations[] ={
        "0n-80s.radionetz.de:8000/0n-70s.mp3",
        "mp3.ffh.de/radioffh/hqlivestream.aac", //  128k aac
};

long t;
int c=0;
void setup() {
    Serial.begin(115200);
    WiFi.begin(ssid.c_str(), password.c_str());
    while (WiFi.status() != WL_CONNECTED){ delay(2000); }
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(10); 
    audio.connecttohost(stations[c].c_str());
    t = millis();
}

void loop() {
    audio.loop();
    if(millis()>(t+7000)) {
      audio.connecttohost(stations[c].c_str());
      c = (c+1) % 2;
      t = millis();
    }
}

void audio_info(const char *info){
    Serial.print("audio_info: "); Serial.println(info);
}
```
Using platform **2.0.17**, it works fine:
```
audio_info: PSRAM found, inputBufferSize: 638965 bytes
audio_info: buffers freed, free Heap: 209840 bytes
audio_info: Connection has been established in 148 ms, free Heap: 208580 bytes
audio_info: MP3Decoder has been initialized, free Heap: 184184 bytes , free stack 6220 DWORDs
audio_info: stream ready
audio_info: syncword found at pos 0
audio_info: Channels: 2
audio_info: SampleRate: 44100
audio_info: BitsPerSample: 16
audio_info: BitRate: 128000
audio_info: StreamTitle='The Rubettes - Sugar Baby Love '
audio_info: Connect to new host: "http://0n-80s.radionetz.de:8000/0n-70s.mp3"
audio_info: buffers freed, free Heap: 208824 bytes
audio_info: Connection has been established in 137 ms, free Heap: 208584 bytes
audio_info: MP3Decoder has been initialized, free Heap: 184188 bytes , free stack 6220 DWORDs
audio_info: stream ready
audio_info: syncword found at pos 0
audio_info: Channels: 2
audio_info: SampleRate: 44100
audio_info: BitsPerSample: 16
audio_info: BitRate: 128000
[... 45 lines cut ...]
audio_info: slow stream, dropouts are possible
audio_info: Connect to new host: "http://mp3.ffh.de/radioffh/hqlivestream.aac"
audio_info: buffers freed, free Heap: 211004 bytes
audio_info: Request http://mp3.ffh.de/radioffh/hqlivestream.aac failed!
audio_info: Connect to new host: "http://0n-80s.radionetz.de:8000/0n-70s.mp3"
audio_info: buffers freed, free Heap: 211024 bytes
audio_info: Connection has been established in 187 ms, free Heap: 210420 bytes
audio_info: MP3Decoder has been initialized, free Heap: 185828 bytes , free stack 6332 DWORDs
audio_info: stream ready
audio_info: syncword found at pos 0
audio_info: Channels: 2
audio_info: SampleRate: 44100
audio_info: BitsPerSample: 16
audio_info: BitRate: 128000
audio_info: StreamTitle='Sweet - Teenage Rampage '
audio_info: Connect to new host: "http://mp3.ffh.de/radioffh/hqlivestream.aac"
audio_info: buffers freed, free Heap: 210632 bytes
**audio_info: Request http://mp3.ffh.de/radioffh/hqlivestream.aac failed!**
audio_info: Connect to new host: "http://0n-80s.radionetz.de:8000/0n-70s.mp3"
audio_info: buffers freed, free Heap: 211024 bytes
audio_info: Connection has been established in 178 ms, free Heap: 210420 bytes
audio_info: MP3Decoder has been initialized, free Heap: 185836 bytes , free stack 6332 DWORDs
audio_info: stream ready
```
Thank you for consideration. Let me know if more details are needed.

shbgr, 2024-05-31:
Arduino 3.0 brings IDF 5 with this new error code 263.
Likely the mass of debug messages causes some stuttering. 54 bytes (4ms @ 115.2kBaud) every 6.5 ms (@44.1kHz) keep the cpu quite busy.
Maybe try again without debug messages. Or patch the lib for the error.
The 263 code is handled in master, but likely not in the release.

rvbc1, 2024-06-25:
Update: I tested it on Arduino versions 3.0.0, 3.0.1, and 3.0.2, and the issue still exists. However, I was testing with the following library dependency:
```
lib_deps = 
	https://github.com/schreibfaul1/ESP32-audioI2S.git#3.0.8
```

on current Version 3.0.11e (commit 0a1674329449581a70c8e8ca7fff0e406b23d820) everything's sound ok. 

```
lib_deps = 
	https://github.com/schreibfaul1/ESP32-audioI2S.git#0a1674329449581a70c8e8ca7fff0e406b23d820
```

closed 2024-08-08 by github-actions[bot]