schreibfaul1/ESP32-audioI2S #760: audio fails in some positions

lidense, 2024-06-03:
I have a webradio using this nice library running on an M5stack Core2 with a small external DAC hooked to an amplified speaker with a cable audio jack 3.5.
This device works well for the dozen or so mp3 or HLS radios stations that I have in my platformio sketch but in certain positions sound stops or runs intemittently with a period of about 1 sec.
If I move the device in a different position it goes back to working. It doesn't seem to be a bad contact because the device and its DAC are held steady in a box.  Wifi signal seems to be sufficient.  
It reminds me of issues like going into some kind of resonance or interference in certain  positions, any idea?
Thanks

lidense, 2024-06-04:
...the terminal gives this error message when the above happens:
[ 80634][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263

schreibfaul1, 2024-06-04:
> [ 80634][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263

I2C is probably used for communication with an external DAC? I have no idea, maybe an ACK problem.

If dropouts occur with HLS, which is quite possible, you can double the DMA buffer. While a new data packet is being fetched and decoded, the DMA must bridge the time. Unfortunately, the DMA in the ESP32 is very small.

lidense, 2024-06-05:
OK, thanks.
Regards

----- Mail d'origine -----

https://github.com/schreibfaul1/ESP32-audioI2S/issues/760#issuecomment-2148316555

lidense, 2024-06-05:
what is the maximum size of dma? 
And of dma count?
Thanks

closed 2024-07-20 by github-actions[bot]