schreibfaul1/ESP32-audioI2S #762: MP3 file cannot be read

AntoineLep, 2024-06-04:
Hi,

I'm currently using the lib to develop an audio book speaker and I've encountered an issue with some files.
The files come directly from an audio book producer and are available on platform such as audible.

Some of the audio books can be read without any problem but for others I can't read it despite being able to do it on a computer.

Here is what the logs tell me when the reading doesn't start:

```
INFO:        PSRAM found, inputBufferSize: 638965 bytes
INFO:        buffers freed, free Heap: 287380 bytes
INFO:        Reading file: "/audiobooks/ed790cc8-8327-4ad2-9d1d-8365c227ca86/3328140024906_02.mp3"
INFO:        MP3Decoder has been initialized, free Heap: 259500 bytes , free stack 5548 DWORDs
INFO:        Content-Length: 37373465
INFO:        ID3 framesSize: 1096908
INFO:        ID3 version: 2.4
INFO:        ID3 normal frames
```

And now when it starts:

```
INFO:        PSRAM found, inputBufferSize: 638965 bytes
INFO:        buffers freed, free Heap: 287380 bytes
INFO:        Reading file: "/audiobooks/ed790cc8-8327-4ad2-9d1d-8365c227ca86/3328140024906_02.mp3"
INFO:        MP3Decoder has been initialized, free Heap: 259500 bytes , free stack 5548 DWORDs
INFO:        Content-Length: 37373465
INFO:        ID3 framesSize: 1096908
INFO:        ID3 version: 2.4
INFO:        ID3 normal frames
ID3DATA:     SettingsForEncoding: Lavf52.64.2
ID3DATA:     Artist: Gisèle Halimi
ID3DATA:     Album: Fritna
ID3DATA:     Track: 1/20
ID3DATA:     ContentType: Livres et histoires
ID3DATA:     Composer: Gisèle Halimi
ID3DATA:     Title: Les yeux noirs, les yeux gris
INFO:        Audio-Length: 37295461
INFO:        stream ready
INFO:        syncword found at pos 0
INFO:        Channels: 2
INFO:        SampleRate: 44100
INFO:        BitsPerSample: 16
INFO:        BitRate: 256000
```

The minimal code I use to test it is:

```c
#include <SD.h>
#include <Audio.h>
#include <SPI.h>
#include <Arduino.h>
#include "driver/i2s.h"
#include <Wire.h>

#define ES8388_ADDR 0x10

#define SD_CS 13
#define SPI_MOSI 15
#define SPI_MISO 2
#define SPI_SCK 14
#define I2S_DOUT 26
#define I2S_BCLK 5
#define I2S_LRC 25
#define I2S_DIN 35
#define I2SN (i2s_port_t)0
#define SDA 18
#define SCL 23

#define PA GPIO_NUM_21   // Amp power ON

Audio audio;

void setup() {
  Serial.begin(115200);

  Wire.begin(SDA, SCL);

  Serial.println("Configuring SPI");
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);

  // Initialize SD card
  if (!SD.begin(SD_CS)) {
    Serial.println("Failed to initialize SD card");
    return;
  }

  // Amp power enable
  gpio_reset_pin(PA);
  gpio_set_direction(PA, GPIO_MODE_OUTPUT);
  gpio_set_level(PA, 1);

  // Initialize I2S
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(8);

  // Start playing MP3 file
  audio.connecttoFS(SD, "/audiobooks/36eabb7a-87f5-4ddb-b41f-53b8ecf4ece7/3328140021905_01.mp3");  // Should be good
  // audio.connecttoFS(SD, "/audiobooks/ed790cc8-8327-4ad2-9d1d-8365c227ca86/3328140024906_02.mp3");  // Shouldn't
}

void loop() {
  if (audio.isRunning()) {
    audio.loop();
  }
}

void audio_info(const char* info) {
  Serial.print("INFO:        ");
  Serial.println(info);
}

void audio_id3data(const char* info) {
  Serial.print("ID3DATA:     ");
  Serial.println(info);
}

void audio_eof_mp3(const char* info) {
  Serial.print("MP3 EOF:     ");
  Serial.println(info);
}

void audio_bitrate(const char* info) {
  Serial.print("BITRATE:     ");
  Serial.println(info);
}

void audio_icyurl(const char* info) {
  Serial.print("ICYURL:      ");
  Serial.println(info);
}
```

- The file that can be read: https://cdn.staytuned.io/misc/share/3328140021905_01.mp3
- The file that can't: https://cdn.staytuned.io/misc/share/3328140024906_02.mp3

If you see anything or have any information to share I'll take it :)

Thanks,

Antoine

schreibfaul1, 2024-06-04:
Hi Antoine,
this is a problem with large embedded images. If you use SD_MMC you will not notice this. Obviously SD (via SPI) takes much longer to recognise and skip the image. I don't have a good solution yet, as a workaround you can increase the waiting time

[image]

AntoineLep, 2024-06-05:
Hi,

Ok I see, thanks for the workaround. It works :).
Do you see a way to skip the image to get rid of the latency involved by the workaround ?

closed 2024-07-03 by AntoineLep