schreibfaul1/ESP32-audioI2S #767: Compatibility with Arduino Core 3.0.1

kungfupizza, 2024-06-13:
ESP32 Arduino Core 3.0.1 has recently come up. Is this library compatible with the same?

tueddy, 2024-06-13:
I successfully compiled the library in PlatformIO with Arduino 3.0.1. 
Sound also came out of the speaker, so the answer is yes.
@kungfupizza So there is no reason not to try it yourself..

kungfupizza, 2024-06-13:
Thank you for confirmation. I did try and I could play audio from SD card but not from HOST. Has that worked for you ? @tueddy

cziter15, 2024-06-21:
I have auto-cpu scaling done in my project for Mp3 playing (jumping from 80M to 240) and it seems that audio plays too fast.

jonnybergdahl, 2024-08-03:
Compiling the internalDAC.ino file with ESP32 Arduino Core 3.0.4 gives the following compile error. 
```
Compilation error: 'I2S_DAC_CHANNEL_BOTH_EN' was not declared in this scope
```

cziter15, 2024-08-03:
> Compiling the internalDAC.ino file with ESP32 Arduino Core 3.0.4 gives the following compile error.
> 
> ```
> Compilation error: 'I2S_DAC_CHANNEL_BOTH_EN' was not declared in this scope
> ```

This flag does not exist on 3.x

The constructor definition is:
`Audio(bool internalDAC = false, uint8_t channelEnabled = 3, uint8_t i2sPort = I2S_NUM_0); // #99`

schreibfaul1, 2024-08-03:
In Arduino V3, IDF V5.1 runs in the background. Unfortunately this version does not recognize an internal DAC regarding I2S. The latest version that supports the internal DAC is Arduino 2.0.17

closed 2024-09-18 by github-actions[bot]