schreibfaul1/ESP32-audioI2S #768: Web radio request failed

ChassaB, 2024-06-14:
Thanks for the great library. Having fun with it. Running happily on an LOLIN S3.
With the web radio sketch I have tried connecting to many of the stations and nearly always get a request failed in the info messages. See attached.

Setting  audio.setConnectionTimeout (500, 3000); seems to fix it for most of them. Is this the right way to fix it or is there something else?
[image]

schreibfaul1, 2024-06-14:
I can't say exactly what happens in the mbedtls area, but 4.6s for an SSL connection is too long. Could it be that the system clock is not 240MHz?

ChassaB, 2024-06-14:
Here are my Arduino IDE settings. I'm in Australia, not that that should make that much difference.
[image]

ChassaB, 2024-06-14:
Here are my IDE settings. Not sure why it should take so long even if I am in Australia.
[image]

closed 2024-07-30 by github-actions[bot]