schreibfaul1/ESP32-audioI2S #774: [ESP32S3 3.0.1] Too fast / high pitch playing - text to speech / google

cziter15, 2024-06-23:
Hello,

I've encountered an issue with TTS on my custom board with ESP32S3 and MAX9835A.

When TTS request is issued, the audio plays too fast and with higher pitch.
Other things like playing from stream or filesystem aren't impacted by the issue.

I'm using 3.0.1 Arduino core.

x9prototype, 2024-06-24:
I have the same boards and same core, and am experiencing the same issue

cziter15, 2024-06-24:
@schreibfaul1 you've referenced a commit that is targeting internal DAC, while this issue is targeting MAX9835A (external DAC)

cziter15, 2024-06-26:
@schreibfaul1 using latest master 0a1674329449581a70c8e8ca7fff0e406b23d820 and still experiencing the issue

NewoPL, 2024-06-30:
I verified that last commit 0a1674329449581a70c8e8ca7fff0e406b23d820 completly distorts the audio from google TTS mp3 
[home.zip](https://github.com/user-attachments/files/16045156/home.zip)

cziter15, 2024-07-02:
This issue is related to bps. I've tried streaming mp3 with various bitrates and everything below 128 was affected.

msw1998, 2024-07-07:
Any fix for this issue?

cziter15, 2024-07-14:
@schreibfaul1 any ideas where to look?

cziter15, 2024-07-18:
This issue seems to be fixed

closed 2024-07-18 by cziter15