schreibfaul1/ESP32-audioI2S #776: MP3 files playback stop

Dcmwrnd, 2024-06-25:
What a great libraries for playing audio files. I did a mp3 player 1 year plus ago using a ESP32, SD card module & the PCM5102A module
for audio output & was working perfectly up to v3.0.8u. When I compile with the latest v3.0.11c or v3.0.11d, it will play the first mp3 file
only than stop playing. Whereas the old version library will play all the mp3 files in the SD card & when I revert back the libraries to
v3.0.8u it will play all.

Basically I saw the audio_eof_mp3 ended message after playing the first mp3 file & it just stop without playing the next mp3 file on the new version.

Regards.

schreibfaul1, 2024-06-26:
In principle, the code would look like this:
````c++
void setup() {
    ...
    ...
    audio.connecttoFS(SD, "/file1.mp3");
}

void loop(){
    audio.loop();
}

void audio_eof_mp3(const char *info){  //end of file
    static int i = 0;
    if(i == 0)  audio.connecttoFS(SD, "/file2.mp3");
    if(i == 1)  audio.connecttoFS(SD, "/file3.mp3");
    if(i == 2)  audio.connecttoFS(SD, "/file4.mp3");
    ...
    i++;
}
````

Dcmwrnd, 2024-06-28:
Thanks for your reply. That was exactly what I did on the ‘void audio_eof_mp3(const char *info)’. In fact I did not alter my code from v3.0.8u. Just recompile with the new version & it does not play all files. With the new version I also notice that when volume button is pressed during playback, the respond is very slow, have to press volume button a few times than only registered.

Tested using Arduino IDE v1.8.19 & ESP32 board manager v2.0.17 on Board: ESP32 Dev Module.
I even tested on the latest ESP32 board manager v3.0.2 with the same result.

Anyway, I can still use v3.0.8u for time being.

Regards.

closed 2024-08-12 by github-actions[bot]

Dcmwrnd, 2024-09-22:
Finally with the latest Audio.h (v3.0.12n) and Audio.cpp (v3.0.12q) and whenever the ‘void audio_eof_mp3(const char *info)’ is triggered, it will continue playing the next assigned mp3 file. Great job!

Regards.