schreibfaul1/ESP32-audioI2S #778: Internal DAC initialization problem

Dread5151, 2024-06-27:
I'm using your "internalDAC.ino" example in my project with an ESP32 Lonin32.  It was working great until around May 2024, when I compile the schetch now it gives the following error:

error: 'I2S_DAC_CHANNEL_BOTH_EN' was not declared in this scope
   11 | Audio audio(true, I2S_DAC_CHANNEL_BOTH_EN);

If I take away the I2S_DAC_CHANNEL_BOTH_EN and just use Audio audio(true); it compiles but then the DAC is not initialized and there is no audio output.
What am I missing?

Regards

schreibfaul1, 2024-06-27:
You must be using Arduino V3...
The last working version is V2.0.17. Arduino V3 uses the IDF V5.1 which according to the I2S specification no longer supports the internal DAC.

Dread5151, 2024-06-28:
AH!!!! Thanx @schreibfaul1 I just couldn't understand why my project was not compiling. I would never have got there.  All good and running now!

Thanx Again!!

closed 2024-08-12 by github-actions[bot]