schreibfaul1/ESP32-audioI2S #784: frying noise

lidense, 2024-07-07:
I have made a webradio using this library.
It works very well with Adafruit feather S3 while using a lolin S3 wroom I hear a very fine frying noise, like very find crackling when sound is heard, not when it is silent. Any idea?

lidense, 2024-07-08:
I'm using a UDA1334A.
Regards

kungfupizza, 2024-07-09:
I had similar issues when I used PINs 0, 1, 2 on ESP32S3. Also could bue due to tangled wires and interference.

Arielhh, 2024-07-10:
sounds like a grounding issue

KartoffelToby, 2024-08-17:
had this issue to with MAX98357 and ESP32 S3, even on other pins

Arielhh, 2024-08-17:
Remove the 2 capacitors on the output stage. Remove also the 2 ferrites on the output stage and replace them with solder short. This will eliminate that noise.

KartoffelToby, 2024-08-20:
@Arielhh thanks for the replying, have you any Pictures of this mod?

I found a other solution, but it still have some noise.

I found the main noise is gone when i connect it to 3,3v on the ESP not 5v direct.

And also make sure if your playing from a SD card its connected via SD_MMC, lack of speed produce noise too

Arielhh, 2024-08-21:
> @Arielhh thanks for the replying, have you any Pictures of this mod?
>
> I found a other solution, but it still have some noise.
>
> I found the main noise is gone when i connect it to 3,3v on the ESP not [...]

send me a photo of your sound module and i will poit you to the right components.
You can also try putting a large electrolytic capacitor  (~470uf or higher)  on the 5V close to the amplifier module.

KartoffelToby, 2024-08-22:
@Arielhh sure no problem, here is a Picture [image]

Arielhh, 2024-08-22:
There are 4 components on the top of the photo.
The two from both out sides left and right are capacitors connected to the ground - remove them
The two middle ones are ferrite remove them and instead replace each one with a short
You can have a look at the electronic schematic provided by Adafruit. 
These are supposed to act as filters but if you look with an oscilloscope you will see they are generation some noise instead of filtering it.

closed 2024-10-06 by github-actions[bot]