schreibfaul1/ESP32-audioI2S #785: mp3 decoder cannot allocate memory

kungfupizza, 2024-07-11:
I have been trying multiple options, with PSRAM, without PSRAM, increasing the stack size for audio device, and more but haven't been successful in getting this error to go away.

`
I (30016) AUDIO: buffers freed, free Heap: 14780 bytes
[ 30049][E][mp3_decoder.cpp:1600] MP3Decoder_AllocateBuffers(): not enough memory to allocate mp3decoder buffers
I (30059) AUDIO: The MP3Decoder could not be initialized, free Heap: 10424 bytes , free stack 9656 DWORDs
`
Any ideas on what I can do to improve it?

Sakuranda, 2024-08-05:
The same thing I got, I used spi 4M flash but there's no way to call it. Just RAM or PSRAM, instead of FLASH

schreibfaul1, 2024-08-05:
SRAM is limited on the ESP32 (~320KB). If your main program has a lot of SRAM, there is not enough left for the decoders. PSRAM would help here, the size of the FLASH does not matter.
One possible option would be to reduce the size of the (ARDUINO LOOP TASK) stack. 
Write `SET_LOOP_TASK_STACK_SIZE(xxxx);` before the setup() and try to reduce xxxx, I think 8192 is the default setting

Sakuranda, 2024-08-05:
> SRAM is limited on the ESP32 (~320KB). If your main program has a lot of SRAM, there is not enough left for the decoders. PSRAM would help here, the size of the FLASH does not matter. One possible [...]

Thanks for your reply, so there's no way to use SPI FLASH to decode?

closed 2024-09-19 by github-actions[bot]