schreibfaul1/ESP32-audioI2S #786: audio.connecttoFS() can't play audio with 8 BITS_PER_SAMPLE ?

kaloprojects, 2024-07-11:
Hi Schreibfaul1, :) 
first of all i want to send you a **huge Thank you**! your library is awesome, using audio.connecttohost() + audio.connecttoFS() daily, also love your Google and OpenAI TSS call. fyi: I referenced to your library on my Github project (I2S Voice Recording & SpeechToText via Deepgram), let me know if you don’t wish to reference to you in my README

To the topic/subject - the issue i found: _Are you aware that your current 'audio.connecttoFS()' does not support audio with real '8-bit width data_ ? Pretty sure it is not your fault, have seen that latest I2S reference is telling that ish_std.h always using 16-bit-with (even if 8BIT defined!). See here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#std-tx-mode. 
This Espessif workflow of course is a pain and makes not much sense. So what i did (in my I2S recording procedure: converting this I2S 'fake' 8BIT with upper msb zeros into real 8 bit waves (and storing with proper wav header). Reason: Pretty simple, 8bit wave files (mono) have half size of 16 bit so any actions SD file read and sending to STT services have double speed. So meanwhile using 8bit only .. but can’t play them any longer with your library

I attached a zip with correct wave files (correct wav header too), both work perfect on PC .. but on ESP32 the 'audio.connecttoFS()' fails with the 8 bit version. Could you pls. check once ? Thank you !

kaloprojects, 2024-07-11:
here to test: 
[Audio_Samples.zip](https://github.com/user-attachments/files/16175363/Audio_Samples.zip)

schreibfaul1, 2024-07-17:
Some external DACs, like the PCM5102A, require 16...32 bits per sample. This is the reason why I always convert 8 to 16 bit. 8 bit mono wav is rarely used, I hope it works better now,

schreibfaul1, 2024-07-18:
I'm not very familiar with the Arduino IDE, I don't know where the libraries are stored by default. My suggestion would be to download the repository as a zip and then unzip the src folder into your sketch folder.

kaloprojects, 2024-07-18:
Haha .. same on my side (so i am not alone on this) , this Arduino IDE is a bit tricky.
i deleted folder .. installed your latest zip file (from today, 3.0.11g with the pitch bug fix too):

_it works PERFECT :), Thank you so much for this work!. I also tested the low resolution wav files (8kHz /8bit mono) they work well and fast for STT services (i tested with Deepgram) .. and your latest fix allows us now to play such tiny wav files also ESP32 I2S DACs. So GREAT. 

Thank you Schreibfaul ! :)

closed 2024-09-01 by github-actions[bot]