schreibfaul1/ESP32-audioI2S #794: Some radio stations continue to have dropouts

renatoianhez, 2024-08-02:
Even after replacing the modified Lolin32 with the ESP32 Wrover, I still have problems with excessive dropouts on some radio stations. Especially stations located in Europe. Examples:
https://servidor33-3.brlogic.com:7064/live
https://s7-webradio.oldie-antenne.de/oldie-antenne
https://italianradio.streamingmedia.it/play
Other stations simply don't play:
https://antenaone.crossradio.com.br/stream/1;
My internet has a speed of 600 Mbps (ping 8 ms), with CloudFire DNS.
Are there any adjustments I need to make to my [code ](https://github.com/renatoianhez/Lolin32-radio/tree/main/RadioWeb)or [hardware](https://github.com/renatoianhez/Lolin32-radio)?

schreibfaul1, 2024-08-03:
I tested the stations with the ESP32-TTGO. No dropouts, the bugger is well filled. Exception: https://servidor33-3.brlogic.com:7064/live the stream is very poor. Tested in the ArduinoIDE with the sketch from the readme.

However, it also depends on how your project is structured. In the newer version, there is a separate task for decoding. Previously, this was done by the "loop" task. The problem was the non-blocking WiFiClient. The WiFi client has the property of waiting for your HTTZ response when "/GET" is sent. This can take longer than there is data in the I2S DMA, resulting in dropouts. The new task has always run on core 0, I will make this adjustable in the future.In principle, the task should always run on the core on which the Arduino Loop Task is not running. Please check whether "Arduino Runs On: Corre1" is set for you.

I have had very good experiences with it so far. Even FLAC with > 1Mbit/s now runs smoothly, like http://s2.audiostream.hu:8091/bdpstrock_FLAC without having to change the WiFi settings in the menuconfig.

Arielhh, 2024-08-21:
what tasks should run on core 0? should I  make audio.connecttohost()  run ob core 0?

schreibfaul1, 2024-08-21:
This has little to do with which core is used. In this case, it helps to increase the WiFi data throughput. 
[menuconfig.pdf](https://github.com/user-attachments/files/16690395/menuconfig.pdf)

Arielhh, 2024-08-21:
Yes, I understand, thanks. I am using Arduino IDE so cannot implement these changes. 
If you have the chance, will you be able to check if the following station can play after implementing the above changes http://87.120.6.86:8000/hq

Arielhh, 2024-08-21:
many thanks, i will try that

Arielhh, 2024-08-24:
I noticed a very strange behavior,  i tried playing the following station and i got dropouts due to low bit rate   http://87.120.6.86:8000/hq I tried to slow the loop by using a temporary delay(20); and the station is working just fine. I can understand that now the buffer can be field with more info before playing but i don't want to use delays in my loop code. 

Is that a normal behavior?

schreibfaul1, 2024-08-27:
Arielhh wrote: 
_I noticed a very strange behaviour, i tried playing the following station and i got dropouts due to low bit rate http://87.120.6.86:8000/hq I tried to slow the loop by using a temporary delay(20); and the station is working just fine. I can understand that now the buffer can be field with more info before playing but i don't want to use delays in my loop code. Is that a normal behaviour?_

I can't reproduce the effect, but I'll try to explain it anyway. The undecoded raw data is first stored in a memory (inbuffer). This is done by audio.loop(). As a rule, audio.loop() is triggered by the Arduino looop(), i.e. it runs in the Arduion LoopTask. The core on which the Arduio LoopTask runs can be set in many IDEs. Within the library there is another task that fetches the data from memory, initiates the decoding and feeds the I2S. This task runs on core 0 by default, but can be changed with `setAudioTaskCore(uint8_t coreID);` Internally, the WiFi processes (probably) run on core 1 for the ESP32 and on cores 1 and 2 for the S3. If the audio.loop() is preceded by a vTaskDelay(), the WiFi process can be given more computing time.

Arielhh, 2024-08-27:
Wolle, thanks for the explanation.  
I will try to play with setAudioTaskCore(uint8_t coreID); to see if there is any improvement.

Arielhh, 2024-08-28:
Actually i don't see much of a difference when switching between core 0 and 1 using setAudioTaskCore(uint8_t coreID); 
all is working fine except of some FLAC stations.

schreibfaul1, 2024-08-28:
yes, the only really good way is to compile Arduino as a component and make the changes in menuconfig.

schreibfaul1, 2024-08-28:
I have updated the example, you first download the repository as a zip and unzip only the folder Better_WiFi_throughput, which you then open in PlatformIO. It will look like this:
 [image]
Then select your board at the bottom, possibly the flash size in platformio.ini and then set your GPIOs in src/main.cpp.

Arielhh, 2024-08-28:
Thanks I will try that.
Maybe it will make it easy for me to understand why my code is not
complying.

Arielhh, 2024-08-28:
i managed to upload the code and use the Run Menuconfig but the controller is keep rebooting. on the terminal looks like that:
[image]
I am using Lilygo t-display S3 it has ESP32-S3 with 16M

closed 2024-10-13 by github-actions[bot]