schreibfaul1/ESP32-audioI2S #796: new 3.012

stef-ladefense, 2024-08-02:
hi,

the latest version 3.012 contains version 3.012e and not 3.012f.
is this normal?

thank,

Stef

schreibfaul1, 2024-08-03:
hi, I rarely create tags, usually after a major change. 
For the very latest version, the only option is to update via Git.

stef-ladefense, 2024-08-03:
thank you
i like this library, i use it to give new life to a 10 year old obsolete web radio. (thanks Orange) by changing its heart.

currently i store the url, the name and a jpg per station line.
does this information exist in the stream?
Is there a way to retrieve the name of the station, its logo, or see information on the title or program directly from the stream (like on DAB)?

closed 2024-09-18 by github-actions[bot]