schreibfaul1/ESP32-audioI2S #800: No audio with JC3636W518 (ESP32-S3 with a 1.8 inch screen and pcm5100a DAC)

tuxfamily, 2024-08-07:
Hello.

I received a new toy, pretty promising, but I'm not able to get any sound using the latest esp32-audioi2s release (3.0.12).

Here is the specs of this board:

[JC3636W518 Specifications-EN.pdf](https://github.com/user-attachments/files/16530237/JC3636W518.Specifications-EN.pdf)

```
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
Manufacturer: 85
Device: 2018
Detected flash size: 16MB
Flash type set in eFuse: quad (4 data lines)
```

And here the schematic diagram of the board and the DAC (apparently a TI pcm5100a chip)

[2_S3R8.pdf](https://github.com/user-attachments/files/16530259/2_S3R8.pdf)

[3_DAC.pdf](https://github.com/user-attachments/files/16530260/3_DAC.pdf)

So, according to this diagram:

[image]

The DAC uses GPIO 16 / 17 / 18 / 21.

I've tried to run the `plays_all_files_in_a_directory.ino` example (without any extra libs or code) using the following PIN config:

```
#define I2S_LRC     16
#define I2S_DOUT    17
#define I2S_BCLK    18

#define SD_MMC_D0   2
#define SD_MMC_CLK  3
#define SD_MMC_CMD  4
```

The mp3 files are correctly found on the card and appear to be playing, but I am not getting any audio output.

When I tried the same mp3 files with the provided (demo) rom, they are well played, so I don't think it's a codec issue.

What I miss? What I'm doing wrong?

Thanks.

schreibfaul1, 2024-08-07:
I would assume that it is a problem with the DAC. PCM5100a is probably rather rare, but according to the datasheet it will work. Maybe set XMST to High?

tuxfamily, 2024-08-07:
Thank you for the suggestion, unfortunately, it does not change a thing.

```
    pinMode(XSMT, INPUT);
    digitalWrite(XSMT, HIGH);
```

(I also tried `LOW` and `audio.setVolume(100)`, just in case)

Here is the console output:

```
Listing directory: /music
  FILE: bayonetta.mp3  SIZE: 3472266
  FILE: qingchun.mp3  SIZE: 1988106
  FILE: 032.mp3  SIZE: 3045983
num files 3playing /music/bayonetta.mp3
info        PSRAM found, inputBufferSize: 638965 bytes
info        buffers freed, free Heap: 284908 bytes
info        Reading file: "/music/bayonetta.mp3"
info        MP3Decoder has been initialized, free Heap: 280504 bytes , free stack 5196 DWORDs
info        Content-Length: 3472266
info        ID3 framesSize: 138
info        ID3 version: 2.4
info        ID3 normal frames
id3data     SettingsForEncoding: Lavf58.76.100
info        Audio-Length: 3472128
info        stream ready
info        syncword found at pos 0

info        Bitsinfo        BitRate: 128000
info        Closing audio file "bayonetta.mp3"
eof_mp3     bayonetta.mp3
playing /music/qingchun.mp3
info        buffers freed, free Heap: 284876 bytes
info        Reading file: "/music/qingchun.mp3"
info        MP3Decoder has been initialized, free Heap: 280476 bytes , free stack 5116 DWORDs
info        End of file "bayonetta.mp3"
```

Everything seems to work as expected, except I don't have any sound or even noise.

This board is only shipped with a (lvgl) sample for the screen along with this PDF:

[Getting started JC3636W518.pdf](https://github.com/user-attachments/files/16532277/Getting.started.JC3636W518.pdf)

But it does not mention a thing about the DAC.

There is also a `pin config` file:

```
#define TFT_BLK 15

#define TFT_RST 47
#define TFT_CS 10
#define TFT_SCK 9
#define TFT_SDA0 11
#define TFT_SDA1 12
#define TFT_SDA2 13
#define TFT_SDA3 14

#define BTN_PIN 0

#define TOUCH_PIN_NUM_I2C_SCL 8
#define TOUCH_PIN_NUM_I2C_SDA 7
#define TOUCH_PIN_NUM_INT 41
#define TOUCH_PIN_NUM_RST 40

#define SD_MMC_D0_PIN 2
#define SD_MMC_D1_PIN 1
#define SD_MMC_D2_PIN 6
#define SD_MMC_D3_PIN 5
#define SD_MMC_CLK_PIN 3
#define SD_MMC_CMD_PIN 4

#define AUDIO_I2S_MCK_IO -1 // MCK
#define AUDIO_I2S_BCK_IO 18 // BCK
#define AUDIO_I2S_WS_IO 16  // LCK
#define AUDIO_I2S_DO_IO 17  // DIN
#define AUDIO_MUTE_PIN 48   // 菴守鳩蟷ｳ髱咎浹

#define MIC_I2S_WS 45
#define MIC_I2S_SD 46
#define MIC_I2S_SCK 42
```

And the spec of the DAC chip itself, here:

[pcm5100a.pdf](https://github.com/user-attachments/files/16531370/pcm5100a.pdf)

But I don't think it's really helpful and I don't know how I can (help you to?) debug this.

tuxfamily, 2024-08-08:
I contacted the manufacturer, and their response was: 
"This is a standard I2S chip, and there are I2S specifications in the data."
... 🤐

The S3 has two standard I2S interfaces, but could there be a conflict with the MIC since it also uses I2S?

Additionally, in the provided sample for the screen, they use the [ESP32_IO_Expander](https://github.com/esp-arduino-libs/ESP32_IO_Expander), so perhaps there is something non-standard in the wiring?

These are just a few leads, as this is not really my area of expertise.

schreibfaul1, 2024-08-08:
Only I2Snum0 is used. I2Snum1 is free for your own applications, such as MIC. The I2S interface is standard.  Your HW may be different somehow. Do you have the possibility to use another DAC? Or a logic analyzer or oscilloscope to see what is happening on the bus lines?
Or is there SW for your HW from the manufacturer to test?

tuxfamily, 2024-08-08:
> Do you have the possibility to use another DAC? 

Nope, no (extra/free) connectors. This is the board:

[image]

> Or a logic analyzer or oscilloscope to see what is happening on the bus lines?

No sorry, I'm just a hobbyist, I don't own such equipment (and knowledge!)

> Or is there SW for your HW from the manufacturer to test?

Unfortunately, no. 
They only provide a code sample for the screen and a bin with a full-featured demo (without the source code). 
Thanks to this bin, I can confirm that the hardware is working (and the files are correctly encoded), but that's pretty much all I can do.

I asked them how to use the DAC, but they simply replied that it's a standard I2S chip, so...

It's really too bad; this board was promising, but without the sound, it lost its appeal (for me).

Anyway, thank you for your time and for looking into this issue.

schreibfaul1, 2024-08-08:
Too bad. I thought the manufacturer would supply sample code with which you can see whether the DAC is working properly. Then you can use the sample software to determine where the problem lies.

tuxfamily, 2024-08-08:
> I thought the manufacturer would supply sample code

Sadly, no, at least not for the DAC, they provide only a partial sample for the screen, and nothing else.

If you want to give a look, this is the provided link/zip with all the PDF and instructions.

http://pan.jczn1688.com/directlink/1/HMI%20display/JC3636W518EN.zip

On a side note, it looks like to be a pretty recent product (released 3 weeks ago).
Hopefully they will improve the doc and the sample in a near future... 🤞🏻

Pako2, 2024-08-12:
@tuxfamily:
Are you sure you tried: `    pinMode(XSMT, INPUT);` ?
Maybe it would be better to use **OUTPUT** instead of **INPUT** !

tuxfamily, 2024-08-13:
@Pako2 Thank you for your suggestion... IT WORKS! 😁 

For me, the `XSMT` is like a "mute button". I assumed it's an `INPUT`, but clearly, it's an `OUTPUT` (even if I don't understand why).

So the trick for this board is simply to add 

```
    pinMode(XSMT, OUTPUT);
    digitalWrite(XSMT, HIGH);
```

in `setup()` (where `XSMT` is defined as `48`)

Thanks a million for helping me sort this out.

closed 2024-08-13 by tuxfamily

Pako2, 2024-08-13:
It immediately caught my eye. XSMT is an INPUT, but that's in terms of the PCM5100A chip. From the point of view of ESP32 (which is the master here), logically this signal must be OUTPUT.
I'm very glad I could help.