schreibfaul1/ESP32-audioI2S #801: Pause and play music

Explorerlowi, 2024-08-07:
How do I pause the music that is playing? After pausing, can I resume the music from where I paused it? If so, how do I do it?

schreibfaul1, 2024-08-08:
There is `audio.pauseResume();` for this

Explorerlowi, 2024-08-08:
[image]
I'm playing the music above.
[image]
Then I executed the pause instruction and called audio.pauseResume(); but the music was still playing and did not pause.
[image]

Explorerlowi, 2024-08-08:
> ![屏幕截图 2024-08-08 131553](https://private-user-images.githubusercontent.com/115329826/356090870-cf751273-7af9-4fad-907b-c78f5433c602.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwOTUzODQsIm5iZiI6MTcyMzA5NTA4NCwicGF0aCI6Ii8xMTUzMjk4MjYvMzU2MDkwODcwLWNmNzUxMjczLTdhZjktNGZhZC05MDdiLWM3OGY1NDMzYzYwMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwOFQwNTMxMjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZWIxYTlkZmM1ODE2M2UwYzU5YTg4ZTdlNzkwNzM4ZjkxYmEwYTliMGI1YmNiZmY1MDBjMjkyYjI5ODFhNjEwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.IefAvp18Wca-CPaLlPMlvQwF0gPGoDhYynIMZM8OdGk) I'm playing the music above. ![屏幕截图 2024-08-08 131610](https://private-user-images.githubusercontent.com/115329826/356090878-ed6d7a3e-bef5-4b84-b706-d3802e43586e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwOTUzODQsIm5iZiI6MTcyMzA5NTA4NCwicGF0aCI6Ii8xMTUzMjk4MjYvMzU2MDkwODc4LWVkNmQ3YTNlLWJlZjUtNGI4NC1iNzA2LWQzODAyZTQzNTg2ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwOFQwNTMxMjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZmE2MDNjYTIzMGMwM2Y4ZTU1NDkwZDdiZjM5OWJhODljZmYxM2JlODhlM2M5ZWQ1NzRiYzdmNWY1NzY5YTZkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.dZrX4E4jH8OIDvp2o5UyjxwJ6bIlQltK6B09UQq7NtY) Then I executed the pause instruction and called audio.pauseResume(); but the music was still playing and did not pause. ![屏幕截图 2024-08-08 131625](https://private-user-images.githubusercontent.com/115329826/356090881-93495dc7-487c-437e-972d-df261cb109c9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwOTUzODQsIm5iZiI6MTcyMzA5NTA4NCwicGF0aCI6Ii8xMTUzMjk4MjYvMzU2MDkwODgxLTkzNDk1ZGM3LTQ4N2MtNDM3ZS05NzJkLWRmMjYxY2IxMDljOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwOFQwNTMxMjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZGVkMDQ0OGU4MmM3ZGNiNzc5YzE2OTBkNTAwNmUyOWJkMzA0MzgxZDA0NDNiNjQ3ZTMzMmQ0OWJkZGY1ZDhkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.25PMGgVWEqgf_gPJKCUWprd6YToObYaatSEa1IHmCDM)

When I use audio2.stopSong();, it can stop the music playing, but I don't know how to resume playing from the paused position

schreibfaul1, 2024-08-08:
perhaps the command is called several times?
`if(audio.isRunning() audio.pauseResume();`

Explorerlowi, 2024-08-08:
I found the reason. When audio.loop() is executed, the m_streamType = ST_WEBSTREAM; set when playing music will be changed to m_streamType = ST_WEBFILE;. As a result, when audio.pauseResume() is called, the judgment of if (getDatamode() == AUDIO_LOCALFILE || m_streamType == ST_WEBSTREAM) does not hold, so the playback cannot be paused and resumed.
[image]

Explorerlowi, 2024-08-08:
> I found the reason. When audio.loop() is executed, the m_streamType = ST_WEBSTREAM; set when playing music will be changed to m_streamType = ST_WEBFILE;. As a result, when audio.pauseResume() is called, the judgment of if (getDatamode() == AUDIO_LOCALFILE || m_streamType == ST_WEBSTREAM) does not hold, so the playback cannot be paused and resumed. ![屏幕截图 2024-08-08 173237](https://private-user-images.githubusercontent.com/115329826/356165328-1f3d05ef-1652-4ae1-9c19-916c6f8ad3fe.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMxMTAxNTcsIm5iZiI6MTcyMzEwOTg1NywicGF0aCI6Ii8xMTUzMjk4MjYvMzU2MTY1MzI4LTFmM2QwNWVmLTE2NTItNGFlMS05YzE5LTkxNmM2ZjhhZDNmZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwOFQwOTM3MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02OTdjZGU4ZTM3MzJhOThhODYxMjRiMGU0MDQxY2M0MDVlNDNkMTU4MzUxYmEzOTc1ODEyOTdkYmQzYTU2ZTE1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.35QInYLg0XyP-w0QzAsjF7p4Hf7r4ZfA_dL5FFaN8Mk)

Change the conditional judgment to if (getDatamode() == AUDIO_LOCALFILE || m_streamType == ST_WEBFILE) to pause and resume playback normally.

schreibfaul1, 2024-08-08:
PauseResume only works with local files. This is not always successful with web files. The server could disconnect if a timeout occurs.

closed 2024-09-22 by github-actions[bot]